stratux/stratux #118: OnOff Shim not installable with eu025 release

Quax-der-Bruchpilot, 2021-06-06:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I just upgraded to the latest version (eu025) and tried to re-install the on-off shim from Pimoroni that I use to shut down the pi and cut power. I downloaded the new stratux version and flashed it via Etcher. I inserted it to my Pi, plugged in the LAN cable and tried to install the OnOff Shim using the bash that Pimoroni provides:

curl https://get.pimoroni.com/onoffshim | bash

However, the installation throws an error saying that the hardware is not supported by the shim. I then reverted to my old eu23 version and it installs just fine. Any idea whats throwing the error here?

b3nn0, 2021-06-06:
Yes. The new version switched to an aarch64 base image - apparently, the software they provide only supports armhf. I'd recommend contacting them and asking for a RaspiOS aarch64 version.
Anyway, since the latest version does pretty much no disk IO any more and writes logs to ramfs by default, it should be pretty safe to just pull the power plug.

Quax-der-Bruchpilot, 2021-06-06:
Thanks for the fast response. I wrote to Pimoroni about this and will post their answer here. In the meanwhile we can consider this as closed!

closed 2021-06-07 by b3nn0