stratux/stratux #119: Stratux does not reastablish communication with OGN

Koudy232, 2021-06-13:
4. Description of your issue:
I've been testing Stratux together with OGN (TTGO T-Beam) as a GPS source. Sometime the communication is interrupted by VHF transmission. If the interruption is long enough (few seconds) the communication does not reestablish, so the position in Stratux is lost but OGN still working - sending its position to the air. I know that is more hardware issue - find better place for Stratux, but still wondering if the communication problem is on Stratux side or the OGN one. I am using EU023 and the OGN firmware from this image. I attache log wile with this event.
[stratux.log](https://github.com/b3nn0/stratux/files/6643858/stratux.log)

b3nn0, 2021-06-13:
This is not something I've seen before, except when my USB cable was dodgy. Apparently, it is impossible to open the virtual serial port created by the Linux kernel for the OGN tracker ("input/output error"). I recommend trying with a different USB cable between the PI and the OGN Tracker.

Koudy232, 2021-06-13:
This Stratux is permanently installed behind baggage compartment relatively close to VHF antenna in fiberglass airplane. I will try better shielding and shorter USB cable. If doesn't help I switch back to internal GPS. I was just wondering if the issue could have software solution because it seems that both Stratux and OGN working, just USB/serial communication is lost. Thank you anyway.

b3nn0, 2021-06-13:
Don't think so. The communication seems to have failed on such a low level that the driver won't allow Stratux to open the serial port any more..

Quax-der-Bruchpilot, 2021-06-13:
I am experiencing the same at the moment. GPS via TTGO but the connection keeps dropping. I then get the message that Serial In is not communicating. I have replaced the TTGO and the GPS antenna already but without success. Have the issue with EU023 and 025 as well. Will try with a new cable now just to check.

b3nn0, 2021-07-19:
Was this issue solved with a different cable?

Koudy232, 2021-07-27:
A have ordered double shielded USB cables, but haven't replaced them yet.

Koudy232, 2021-09-12:
I've replaced the cables with 15cm triple shielded ones. After two weeks of testing seems that communication works with no issues at all even with installations close to VHF antennas.

DocTopi, 2021-09-13:
My solution for this problem: I`ve wired the Raspberry Pi Pins "GND" and "5V+" to the equivalent TTGO T-Beam pins.
Now the connection seems to be stable.
(is it possible that an energy saving function for the USB port causes this problem?)

mrabans, 2021-09-14:
Had the same issue with a 30cm thin and flexible USB cable resulting in about 50 connection losses and reconnections using Skydemon in a 1:30 hours flight. A shielded 15mm USB cable fixed it so far, not a single connection loss recorded after the replacement.

b3nn0, 2021-09-17:
Since this isn't a software issue, I couldn't ever close this. It might be useful if it is found by people, so I'll convert this to a discussion - makes more sense there I guess.

closed 2021-09-17 by b3nn0