stratux/stratux #120: External gps mouse is ignored when using it with a ttgo t-beam for altitude and FLARM emitting

danielreiser, 2021-06-15:
1. Stratux version: v1.6r1-eu025
2. Stratux config: Default

    SDR
      - [ ] single
      - [X] dual
    type:  NooElec Dual-Band NESDR Nano 3 Premium
    
    GPS
      - [X] yes
      - [ ] no
      type:  BL1 External USB GPS module compatible kit

    AHRS
      - [ ] yes
      - [X] no

    power source: anker usb power outlet

    usb cable: regular power cable from raspberry pi

3. EFB app and version: latest SkyDemon
    EFB platform: iOS 14.x
    EFB hardware: iPad Mini 5

4. Description of your issue:

My intention is to build a stratux eu in combination with a ttgo t-beam for emitting FLARM.
To get a better altitude I've soldered a BMP280 to the ttgo t-beam. As the gps wasn't working correctly on the ttgo (no gps fix) and the reception is better on an external antenna anyways, I've bought the external gps mouse. 

The issue I am currently facing is that the settings window is not showing any OGN config, the gps can't get a fix and the baro alt is being used (digging through the code I suppose its not the actual BMP280 being used but rather sth. else 🤷‍♂️). Therefor my assumption is that the external gps mouse is ignored entirely. Is there any way to make the above scenario work? Or should I rather put the BMP280 directly onto the raspberry pi and recompile the esp-ogn-tracker firmware with disabled gps?

Thanks a lot for all the effort you're putting into stratux eu!

Best regards from Hamburg
Daniel

VirusPilot, 2021-06-15:
Hi Daniel,

first of all I would not give up so quickly on the T-Beam GPS - the included little patch antenna is really crappy, this is why some of us have replaced it with a larger patch antenna or even an external antenna, connected via an U.FL-SMA adapter cable. Keep in mind that the U.FL connector is very sensitive and may also be a source for difficulties.

What is strange though is that the OGN config does not show up, did you already flash the OGN tracker firmware? Btw, if you want to really use the T-Beam for emitting FLARM, you have to rely on SoftRF which has quite some drawbacks as you can see from the Stratux Wiki.

Did your separate USB GPS work with Stratux with the T-Beam disconnected? Stratux may use the T-Beam GPS when connected with priority but this question goes to Adrian.

Best Regards from Würzburg,

VirusPilot

PS: you may consider joining our slack channel #stratux-europe-edition (in stratuxadsb.slack.com)

danielreiser, 2021-06-15:
Hey VirusPilot,

thanks for your answer and thanks for the note regarding slack :) Is there any antenna in particular you'd recommend?
I've flashed the tbeam via the bash script on the stratux image. If I don't connect the tbeam but only the external gps, everything works as it's supposed to work (no baro alt cause the sensor is soldered to the tbeam).
Seems that I got some parts wrong with the tbeam then. So just to be sure: the tbeam can be used for emitting FLARM only via SoftRF (which has some drawbacks which I also read about in the wiki), therefore the only usecase for the tbeam in germany is using it as a gps and optionally ahrs and baro module?

b3nn0, 2021-06-15:
Yes, as the Wiki says, the OGN Tracker firmware will transmit OGNTP and PilotAware protocols in parallel. SoftRF can be configured to transmit one protocol of choice.
Anyway yes, Stratux can only handle one NMEA source and will only connect to one such source. In this case, that will be the normal GPS. I.e. your T-Beam will not be connected to, therefore no settings, no pressure data.
I did some work towards supporting multiple sources, but it is far from finished and not sure if I will ever finish it.

But as VirusPilot said, the GPS Chip on the T-Beam is decent, it's just the antenna that's crappy.

Personally, I use this antenna and it works very well: https://www.amazon.de/gp/product/B06WV7CYXS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
but anything that fits your form factor should be better than what comes with the T-Beam.

closed 2021-06-15 by danielreiser