stratux/stratux #128: UAT / 1090 conflict

TrashHauler, 2021-08-15:
1. Stratux version:1.6r1-eu025
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:GPYes 2.0

    AHRS
      - [ ] yes
      - [x ] no

    power source:
Wall power or battery
    usb cable:
Multiple
3. EFB app and version: (e.g., WingX Pro7 8.6.2)
N/A
    EFB platform: (e.g., iOS 9.2)
iOS 14.6
    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
UAT does not receive towers when both UAT and 1090 dongles are attached and active.

Build: rPi 4B using latest EU Stratux version. Stratux v3 UAT plus 1090es with high gain antennas and GPYes2.0 all purchased from Amazon.

Configuration:  OGN off, UAT, 1090, GPS enabled in settings and dongles installed. Result: GPS and 1090 work, UAT no reception. Stratux main page recognizes 2 SDRs but 0 towers and no data received on UAT.

Disable 1090 in settings and UAT receives towers and downloads data. Same result if 1090 dongle is removed and disabled in settings. 

Remove UAT dongle and 1090 works. So both dongles work, they just don’t work together. 

Power is stable and cpu temps around 40C.

I suspect there is a configuration conflict but have not had time to further troubleshoot. Any suggestions on where to start?

I forgot to bring my other SD card with the US software version and have no way to flash a card for several more days so I can’t test if this is a problem only with the EU version or not.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2021-08-15:
Not seen this behavior before. Mind posting a log file?

TrashHauler, 2021-08-15:
It will be tomorrow before I can mess with it again. I did have a thought I want to test out. The 1090 does not have a specific serial number assigned. Where the v3 UAT is technically not an SDR, I wonder if the system is confused and trying to make both dongles UAT? Problem with that theory is that 1090 works so I don’t know. I’ll test it tomorrow and report back.  

Bob

TrashHauler, 2021-08-15:
Played around a bit and here is what I discovered:
1090 dongle does need to be assigned a proper serial number
UAT v3 Serbian number can not be changed using the rtl_eeprom -s command

With these changes, both UAT and 1090 work simultaneous but UAT messages are still relatively low. If I disable 1090 in settings, UAT message count jumps significantly. Re-enable 1090 and UAT messages drop. Limitation of the system? Processing power? This is on a rPi 4B with 4gb ram.

Another problem (this is not a new problem) is with the GPYes 2.0 calculating a solution. It is tracking between 10 and 21 satellites but after 38 minutes of operation, I still do not have a location. I manually updated the time and date on the Pi and rebooted prior to the current 38+ minutes of up time. Time is set to UTC and is within 60 seconds of what gps time should be. No idea why it takes so long for the gps to capture a location. Any suggestions on either issue (UAT messages low with 1090 on or gps not capturing location)?

Current log to follow.

TrashHauler, 2021-08-15:
[stratux.log](https://github.com/b3nn0/stratux/files/6988876/stratux.log)

TrashHauler, 2021-08-15:
Stratux has been up for an hour and 20 minutes and still no gps location. But is sees plenty of satellites.
[image]

b3nn0, 2021-08-16:
Yes specifying serial number seems to be needed in this case.. the algorithm could certainly be improved, but that's just how it is right now.
So I don't see a software issue any more now. Since the UAT radio is no RTLSDR, rtl_eeprom will not work of course - that's as expected.

Processing power is certainly not an issue. The PI4 has far more than enough. and note that setting the time is not needed. The GPS will not know the system time of the PI and works independently.

There are only two things I can imagine:
- Not enough power to reliably power both at the same time/including GPS. Make sure that the **red LED on the PI is on constantly**. If it is not, voltage is too low - try a different power supply or cable then.
- EMI. Sadly, the noise characteristics of the RTLSDRs are quite unpredictable and vary from device to device (not from model to model - really from unique device to another). I've seen some that had no impact at all on GPS, and I've seen some that caused it to instantly lose all satellites and give up. Maybe you were just very unlucky with  your 1090ES SDR..
Shielding can help in this case of course. Does the GPS also not work if the 1090ES SDR is disconnected?

Looking at your log:
Interestingly, while your GPS does seem to have some satellite signals, it doesn't seem to know their location, i.e. not all data is received yet.. no idea why - probably one of the above reasons?

Another note:
You will probably not want to have replay logs enabled in production. It writes a **lot** of data to an SQLite DB (as in multiple hundred MB per hour), which can quickly fill up your SD card, and potentially could cause filesystem corruption when the PI loses power.
Debug logging should also be disabled once everything works.

VirusPilot, 2021-08-16:
All your GPS satellite signals are way below 30dB, there seems to be a significant EMI source that prevents the GPS from an initial fix. Typically such signals are between 30 and 50 dB. What GPS are you using? Can you temporarily use an USB extension cable to move the GPS away from the Pi4?

TrashHauler, 2021-08-17:
Thanks All for the suggestions. GPS worked beautifully in the aircraft today. No issues at all, go figure. 1090 and UAT still have issues but are definitely better with a serial number assigned to the 1090. At 36,000 feet today, I was receiving 2 towers at best on UAT, better with 1090 off but still not great. This is a work in progress and really what I want working is weather radar via UAT. Is there any way to modify the code to prioritize loading radar data? METAR, TAF, PIREPS, etc all seem to load before the radar if it loads at all. 

As for power, that hasn’t been a problem. Solid red on the Pi at all times. 

EMI could very well be the leading problem. Between the SDR, heated windscreens, etc I could be fighting a losing battle. 

For troubleshooting purposes, I set the log to capture everything. I will definitely cut the logging down to minimums once I figure this thing out. Other people I fly with use commercial options and they work great in the aircraft. I had the Pi and much prefer open source solutions so I hope I can get the issues worked out and have weather radar to supplement the aircraft radar for dodging thunderstorms.

b3nn0, 2021-08-17:
Alright, I'll close this then, since it's not a software issue. I assume another issue could be that at 36,000ft, I assume your antennas are inside an airtight tin can - which block RF signals quite a bit.

closed 2021-08-17 by b3nn0