stratux/stratux #131: Suspect memory leak in gen_gdl90

scsirob, 2021-09-01:
1. Stratux version: v1.6r1-eu25
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x ] no

    power source: Fixed 3A USB wall adapter

    usb cable: High-current. No power dips are detected

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2) EasyVFR 4

    EFB hardware: (e.g., iPad Mini 2) Win10 Desktop 

4. Description of your issue: I have a fixed install with an RPi4 at home, feeding into my Ethernet network. I have one designated EasyVFR desktop assigned in the Stratux static IP settings. After a long uptime (> 1 week) gen_gdl90 consumes a large amount of memory (>60%). This triggers the Linux out-of-memory killer. After a reboot memory consumption drops to 3%.
5. Note: I have both FR24 and FA feeder running on the same RPi4, taking their input from dump1090. As gen_gdl90 does not talk to either, I think this is an isolated memory leak in gen_gdl90 code. Not sure if this is in any way related, but I do see traffic listed with -999dB signal strength and some of the entries show a lighter blue icon with what appears to be a satellite dish instead of a plane.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2021-09-01:
About the light blue targets: These are most likely MLAT targets that are back-fed into dump1090 from FR24 or FA.

About the memory leak: Yes, I know about it, it's been there forever and I already fixed a couple of those, but I don't know where this one (or multiple?) is. Since it's small enough to not be relevant for Stratux's actual use case, I didn't bother too much. If someone wants to try and track that down, PR is welcome.

b3nn0, 2021-09-02:
I think I have actually just fixed the issue, at least if "Replay Logs" are disabled. The case where they are enabled is a bit harder to fix because the code is so highly asynchronous.. Will have to do with this for now, since running for multiple days with replay logs enabled is kinda impractical anyway.

closed 2021-09-02 by b3nn0

scsirob, 2021-09-09:
Just to confirm, I applied the replay log patch from eu26 to my eu25 install and it fixed the memory leak for me.
I do not write go, but a quick search on possible memory leak scenarios came up with a hint to time.Ticker instances that need to be stopped to allow garbage collection. As the affected routine branches out to multiple time sources, this may be where the leak is?

b3nn0, 2021-09-09:
No, it's the simple fact that `dataLogTimestamps` is appended to indefinitely, for each GPS signal. And never cleaned up.
We don't need it when replay log is disabled, so the workaround does the job.
However, I couldn't come up with a simple solution to clean up old entries when replay log is enabled, because its super asynchronous and old indices are needed for some time before they can be evicted. But evicting old elements would change the indices of newer elements, .....
Doesn't matter too much IMO.