stratux/stratux #144: T-beam not communicating

mave330, 2021-11-06:
1. Stratux version: Stratux Europe Edition 1.6r1-eu026 on RPi 4
2. Stratux config: 

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: T-beam TTGO T22 V1.1 (lora32

    AHRS
      - [ ] yes
      - [x] no

    power source: USB C power 5V 3A

    usb cable: nimaso usb c to c

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: 
It seems my T-beam is not recognized by stratux. 
I am seeing in stratux interface : Serial port (Not communicating) , GPS solution: Disconnected

- I have previously flashed the t-beam on windows with OGN-Tracker (no error in flash tool), and the t beam was working out of the box with soft rf on the direct web interface of the ttgo- gps aquisition ok.
- I have tried with other USB cable, same behaviour.

Can the problem be from my ttgo HW version ? 

Thanks for your help! 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log
[stratux.log](https://github.com/b3nn0/stratux/files/7490977/stratux.log)
[image]
[image]

.

b3nn0, 2021-11-06:
See here: https://github.com/cyoung/stratux/issues/842

mave330, 2021-11-06:
thanks for your answer!  I'm able to see that I also have the QinHeng Electronic using lsusb.

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 004: ID 1a86:55d4 QinHeng Electronics 
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

However, I cant figure out to find this : /etc/udev/rules.d/10-stratux.conf exist... I'm not a pro using ssh ...

I'll keep searching..

b3nn0, 2021-11-06:
You can also install this beta version that will fix the issue until the final one is released: https://abatzill.de/stratux/stratux-v1.6r1-eu026-f7a4eb4f.img.zip

closed 2021-11-06 by b3nn0