stratux/stratux #152: OpenFlightMap Europe: missing Western Europe ?

MisterT2, 2021-12-19:
1. Stratux version: v1.6r1-eu027
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [ ] yes
      - [X] no
      type:

    AHRS
      - [ ] yes
      - [X] no

    power source: AC converter

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I wanted to test the possibility to add new offline maps
So I followed the recommendations stated here: https://github.com/b3nn0/stratux/wiki/Downloading-better-map-data, and I downloaded OpenFlightMap Europe

I can activate this map (instead of OSM one), but I'm missing all western part of Europe...
Did I missed anything ?
[image]

Thanks a lot !

b3nn0, 2021-12-19:
Yes you are correct. Sadly, these are not available for download in OFM yet: https://www.openflightmaps.org/
there is no France/Spain.. hopefully that's coming at some point. Nothing I can do about it.

closed 2021-12-19 by b3nn0

MisterT2, 2021-12-19:
Obviously nothing you can do there. I will check on OpenflightMaps then...
Thanks for your fast answer....