stratux/stratux #161: Stratux runs out of memory

dd5fz, 2022-02-19:
1. Stratux version: Stratux v1.6r1-eu027 / Linux version 5.10.63-v8+ / Raspberry Pi 3 Model B Plus Rev 1.3

2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual (NESDR Nano 2)

    GPS
      - [ x] yes
      - [ ] no
      type: USB u-blox 8 GNSS receiver

    AHRS
      - [x ] yes, but non-functional (!)
      - [ ] no

    power source: 230V/5V power supply - test setup at home
    usb cable: A standard cable to supply power

3. EFB app and version: EasyVFR
    EFB platform: 15.3.1
    EFB hardware: iPad Mini (5th generation)

4. Description of your issue:

 Due to a problem with the AHRS/air pressure sensor board and/or the i2c bus on the Raspberry Pi, many thousands of lines of error messages are being logged, e.g. in syslog:

Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 Error identifying IMU: remote I/O error
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)
Feb 19 02:04:15 stratux gen_gdl90[474]: 2022/02/19 02:04:15 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)

After a couple hours, the Stratux runs out of space (Free Storage: 0.0 MiB):

-  System uptime 26306 seconds. Daemon was restarted.
-  AHRS Error: Couldn't read pressure from sensor: Pressure is 0
-  Disk bytes used = 259 MB (98.9 %), Disk bytes free = 3.0 MB (1.1 %)
-  Write error on '/var/lib/misc/', your EFB may have issues receiving weather and traffic.
-  Barometric altitude 9942' out of expected range. Ignoring. Pressure sensor potentially broken.

b3nn0, 2022-02-19:
Uhm.. disable the sensors then? They are obviously broken..?

dd5fz, 2022-02-19:
Uhm... yes, I know there is something wrong with the sensors.

However, with the read only FS and writing the logs to RAM (?), I feel that running out of space whether after 2 hours or several days shouldn't be a thing, even if it is, with fully with functional hardware, a somewhat unlikely scenario.

peepsnet, 2022-07-11:
root@stratux:/etc# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  1.4G   27G   5% /overlay/pivot
devtmpfs        422M     0  422M   0% /dev
none            455M     0  455M   0% /overlay/pivot/overlay
none            250M  201M   50M  81% /overlay/rwdata
overlay         250M  201M   50M  81% /
tmpfs           455M     0  455M   0% /dev/shm
tmpfs           182M  5.3M  177M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   31M  222M  12% /boot
tmpfs            91M     0   91M   0% /run/user/1000

b3nn0, 2022-07-12:
> root@stratux:/etc# df -h Filesystem Size Used Avail Use% Mounted on /dev/root 30G 1.4G 27G 5% /overlay/pivot devtmpfs 422M 0 422M 0% /dev none 455M 0 455M 0% /overlay/pivot/overlay none 250M 201M 50M 81% /overlay/rwdata overlay 250M 201M 50M 81% / tmpfs 455M 0 455M 0% /dev/shm tmpfs 182M 5.3M 177M 3% /run tmpfs 5.0M 4.0K 5.0M 1% /run/lock /dev/mmcblk0p1 253M 31M 222M 12% /boot tmpfs 91M 0 91M 0% /run/user/1000
> 
> I am not able to perform an apt update because there is not enough free space.
> 
> I am not able to expand the FS

That is more or less unrelated. See https://github.com/b3nn0/stratux/wiki/Read-only-filesystem

commit 42ba51e6 (2023-05-18): implement custom log-rotation, 10mb max per file, 10 files max, including free space monitoring. Disable stratux logging to syslog/journald. Closes #161
closed 2023-05-18 by b3nn0