stratux/stratux #162: OGN stay disconnected

Mariko64, 2022-04-03:
1. Stratux version: Stratux v1.6r1-eu027
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type: USB u-blox 7 GNSS

    AHRS
      - [ ] yes
      - [ ] no

    power source: powerbank

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: After some successfull  turnON/turnOFF where everything works correctly, the OGN remains disconnected. The only way I have found to have OGN receiving back to work, is to format the SD and install system again, but after some uses, the issues come back.
[image]

2021/10/30 13:43:15  - stratux-pre-start.sh - Running Stratux Updater Script.
2021/10/30 13:43:15  - stratux-pre-start.sh - Exited without updating anything...
2021/10/30 13:43:15  - stratux-wifi.sh - Running Stratux WiFI Script.
2021/10/30 13:43:15  - stratux-wifi.sh - Killing wpa_supplicant AP services 
2021/10/30 13:43:15  - stratux-wifi.sh - Stopping DHCP services 
2021/10/30 13:43:16 can't read settings /boot/stratux.conf: invalid character 'à' looking for beginning of value
2021/10/30 13:43:16 Stratux v1.6r1-eu027 (3d07b461d7a9565cea07727c44760d3b55bb0039) starting.
2021/10/30 13:43:16 Developer mode set
2021/10/30 13:43:16 Entered uatReader() ...
2021/10/30 13:43:16 getDHCPLeases(): open /var/lib/misc/dnsmasq.leases: no such file or directory
2021/10/30 13:43:18  - stratux-wifi.sh - Restarting DHCP services
2021/10/30 13:43:20 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:24 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:28 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:32 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:36 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:40 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:44 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:46 stats [started: 30 seconds ago]
2021/10/30 13:43:46  - Disk bytes used = 238 kB (0.1 %), Disk bytes free = 262 MB (99.9 %)
2021/10/30 13:43:46  - CPUTemp=25.31 [23.36 - 25.80] deg C, MemStats.Alloc=1.6 MB, MemStats.Sys=14 MB, totalNetworkMessagesSent=0
2021/10/30 13:43:46  - UAT/min 0/0 [maxSS=0.00%], ES/min 0/0, Total traffic targets tracked=0
2021/10/30 13:43:46  - Network data messages sent: 0 total.  Network data bytes sent: 0 total.
2021/10/30 13:43:46  - Last GPS fix: 30 seconds ago, GPS solution type: 0 using 0 satellites (1/1 seen/tracked), NACp: 0, est accuracy 999999.00 m
2021/10/30 13:43:46  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 999999 m
2021/10/30 13:43:46  - Mode-S Distance factors (<5000, <10000, >10000): 2500.000000, 2800.000000, 3000.000000
2021/10/30 13:43:46 - Last IMU read: 30 seconds ago, Last BMP read: 30 seconds ago
2021/10/30 13:43:46 client connected: [192.168.10.17:4000](http://192.168.10.17:4000/) ().
2021/10/30 13:43:46 client connected: [192.168.10.17:2000](http://192.168.10.17:2000/) ().
2021/10/30 13:43:46 client connected: [192.168.10.17:49002](http://192.168.10.17:49002/) ().
2021/10/30 13:43:48 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:52 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:43:56 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:44:00 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:44:04 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:44:08 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:44:12 Error identifying IMU: open /dev/i2c-1: no such file or directory
2021/10/30 13:44:16 stats [started: 1 minute ago]
2021/10/30 13:44:16  - Disk bytes used = 242 kB (0.1 %), Disk bytes free = 262 MB (99.9 %)
2021/10/30 13:44:16  - CPUTemp=27.26 [23.36 - 27.26] deg C, MemStats.Alloc=2.1 MB, MemStats.Sys=14 MB, totalNetworkMessagesSent=46
2021/10/30 13:44:16  - UAT/min 0/0 [maxSS=0.00%], ES/min 0/0, Total traffic targets tracked=0
[... 364 lines cut ...]
2022/04/03 16:05:58 Error identifying IMU: open /dev/i2c-1: no such file or directory
2022/04/03 16:05:58 OGN: restarting crashed ogn-rx-eu
2022/04/03 16:05:58 Entered OGN read() ...
2022/04/03 16:05:58 OGN: Executed ogn-rx-eu successfully...
2022/04/03 16:05:58 OGN: ogn-rx-eu stderr:  Unable to open SPI device: No such file or directory
2022/04/03 16:05:58 OGN: ogn-rx-eu stdout:  Aircraft ID: 8:3:5664F4
2022/04/03 16:05:58 OGN: ogn-rx-eu stdout:  APRS client: user OGN5664F4 pass 17632 vers Stratux
2022/04/03 16:05:58 OGN: ogn-rx-eu terminated...
2022/04/03 16:06:02 Error identifying IMU: open /dev/i2c-1: no such file or directory
2022/04/03 16:06:03 OGN: restarting crashed ogn-rx-eu
2022/04/03 16:06:03 Entered OGN read() ...
2022/04/03 16:06:03 OGN: Executed ogn-rx-eu successfully...
2022/04/03 16:06:03 OGN: ogn-rx-eu stderr:  Unable to open SPI device: No such file or directory
2022/04/03 16:06:03 OGN: ogn-rx-eu stdout:  Aircraft ID: 8:3:5664F4
2022/04/03 16:06:03 OGN: ogn-rx-eu stdout:  APRS client: user OGN5664F4 pass 17632 vers Stratux
2022/04/03 16:06:03 OGN: ogn-rx-eu terminated...
2022/04/03 16:06:06 Error identifying IMU: open /dev/i2c-1: no such file or directory
2022/04/03 16:06:08 OGN: restarting crashed ogn-rx-eu
2022/04/03 16:06:08 Entered OGN read() ...
2022/04/03 16:06:08 OGN: Executed ogn-rx-eu successfully...
2022/04/03 16:06:08 OGN: ogn-rx-eu stderr:  Unable to open SPI device: No such file or directory
2022/04/03 16:06:08 OGN: ogn-rx-eu stdout:  Aircraft ID: 8:3:5664F4
2022/04/03 16:06:08 OGN: ogn-rx-eu stdout:  APRS client: user OGN5664F4 pass 17632 vers Stratux
2022/04/03 16:06:08 OGN: ogn-rx-eu terminated...
2022/04/03 16:06:10 Error identifying IMU: open /dev/i2c-1: no such file or directory

b3nn0, 2022-04-03:
First thing that is strange is this:
`2022/04/03 16:06:02 Error identifying IMU: open /dev/i2c-1: no such file or directory`
Do you have an AHRS module connected? If not, you will want to disable AHRS and/or Baro in the settings so Stratux stops trying (and logging) to connect to it.

Second strange thing is this:
2022/04/03 16:06:03 OGN: ogn-rx-eu stderr: Unable to open SPI device: No such file or directory

It kinda seems like your I2C or SPI bus is gone for some reason.. did you modify your /boot/config.txt? or can you post it?
Which SD card model are you using?

Mariko64, 2022-04-03:
Thanks for you help

When I try to disable baro and ahrs, on the main page Stratux shows an error because config file is a read only file.

I tried two SD, both same type, 32GB, purchased on Amazon https://www.amazon.it/gp/product/B00KOGHQZM/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

My config.txt file:
[config.txt](https://github.com/b3nn0/stratux/files/8404978/config.txt)

b3nn0, 2022-04-03:
This config file is completely corrupted and not readable.
I'm pretty sure your SD card is a cheap chinese knockoff, advertising a wrong size. And once some data is written to it, it will overwrite some random other data (in this case, the config.txt file and probably more), breaking the whole system.
I recommend buying a decent brand SD card and your problem will most likely be gone (Sandisk, Samsung or similar).

Mariko64, 2022-04-03:
I think, and hope, you are right! I just ordered a brand new Samsung EVO plus. I will inform you if the issue will be resolved.
Thank you

Mariko64, 2022-04-07:
Good news
After three days of tests, I am pretty sure that the issue was resolved. I used branded microSD and enlarged the case (self made with a FDM printer) window where the microSD has to be slided in the Raspberry.
Thank for assistance!

closed 2022-04-07 by b3nn0