stratux/stratux #168: WiFi AP+Client mode connection problems

mpotthoff, 2022-05-22:
1. Stratux version: 1.6r1-eu027
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: TTGO T-Beam

    AHRS
      - [ ] yes
      - [x] no

    power source: 2A

3. EFB app and version: Skydemon

    EFB platform: Android 12

    EFB hardware: Samsung Active Tab 3

4. Description of your issue:

As the weather is getting better I have decided to unpack my stratux and play around with it. I know that the WiFi AP+Client mode was working perfectly fine when I implemented multiple client networks last year. But for some reason I can now only connect to the stratux AP sporadically when stratux is connected to a network as a client at the same time.

I'm pretty sure I have tracked the issue down to `ap_scan=2` in `wpa_supplicant_ap.conf`. It also causes wpa_supplicant to spit out a warning:
```
Note: nl80211 driver interface is not designed to be used with ap_scan=2; this can result in connection failures
```
(Which seems to describe exactly what I am experiencing)

I have been testing a custom image with `ap_scan` set to `1` lately and it seems to work reliably now.
Do you remember whether there was a specific reason why you configured it to `2`? Because otherwise I would create a PR which changes it to `1`.

b3nn0, 2022-05-22:
I do not remember why I did that, unfortunately, but it was in there since quite early on. But I do also see that the connection isn't too reliable, especially when trying to connect multiple clients to the Stratux in AP+Client mode.
If it is reproducably better with ap_scan=1, then go ahead and make a PR. I'll do some testing, too.

mpotthoff, 2022-05-22:
And just to describe what kind of problems I have been experiencing: I wasn't able to connect at all to the network. My android clients showed that they were trying to connect for around a second and then just stopped. Sometimes a toast message appeared along the lines of "Connection failed". Sometimes a reboot helped, sometimes not.

commit 6dd33047 (2022-05-22): Change ap_scan to 1 to make the connection in AP+Client mode more reliable (Fixes #168)
closed 2022-05-22 by b3nn0