stratux/stratux #185: panic: runtime error: index out of range [3] with length 1

VirusPilot, 2022-07-21:
here is what happens all 20-30s:

2022/07/21 08:35:55 OGN: ogn-rx-eu stdout:  Open RTLSDR[1]
2022/07/21 08:35:57 ogn-rx-eu successfully connected
2022/07/21 08:36:02 Parsing OGN device db
2022/07/21 08:36:03 Successfully parsed OGN device db
**panic: runtime error: index out of range [3] with length 1**

goroutine 43 [running]:
encoding/binary.bigEndian.Uint32(...)
	/root/go/src/encoding/binary/binary.go:112
main.importOgnTrafficMessage({{0x400030e540, 0x3}, 0x62d8f3e0, {0x400030e543, 0x6}, 0x3, {0xca70d8, 0x1}, {0x0, 0x0}, ...}, ...)
	/root/stratux/main/ogn.go:166 +0xc5c
main.aprsListen()
	/root/stratux/main/ogn-aprs.go:232 +0xa20
created by main.initTraffic
	/root/stratux/main/traffic.go:1716 +0xe4
2022/07/21 08:36:21  - stratux-pre-start.sh - Running Stratux Updater Script.
2022/07/21 08:36:21  - stratux-pre-start.sh - Exited without updating anything...
2022/07/21 08:36:21 read in settings.

closed 2022-08-28 by b3nn0

susanneeichel, 2022-08-28:
`EDIT: a switch to enable/disable APRS would be appreciated`
I added the switch to the settings page, in the hardware section:
Internet (OGN APRS)

VirusPilot, 2022-08-28:
perfect!