stratux/stratux #186: adding additional PWR Energy Savings

rudiratlos, 2022-07-27:
I've wrote a script that uses some tweaks to bring power consumption further down.
Maybe you will find it usefull and use parts of the code (enable_pwrsave) directly in your startup script. 
It's only:
  /usr/bin/tvservice -o > /dev/null
  /usr/bin/cpufreq-set -g powersave
additionally you have to:
1. install the sw-package cpufrequtils
2. add some parameters to /boot/config.txt
this is done by calling the script with parameter setup
(pls. see function setup_env)

calling the script with parameter status:
this will show the active cpu freq. set by the cpu governor

e.g.
current CPU frequency is 600 MHz (asserted by call to hardware).

have fun
regards Stefan

`#/bin/bash
# V1.0 by SF 2022-07-27
# inspired by: https://buyzero.de/blogs/news/raspberry-pi-strom-sparen-tipps-tricks

function setup_env {
  fn="/boot/config.txt"
  tm="# define minimum freq. for energy saving (used by cpu governor)"
  grep -Fq "$tm" "$fn"
  _ret=$?
  if [ "$_ret" != "0" ]; then
    if [ "$1" != "noswpkg" ]; then
      apt-get -qy update
      apt-get -qy install cpufrequtils
    fi
    echo " " >> $fn
    echo "# Disable the PWR LED (save energy)" >> $fn
    echo "dtparam=pwr_led_trigger=none" >> $fn
    echo "dtparam=pwr_led_activelow=off" >> $fn
    echo "$tm" >> $fn
    echo "# use turbo speed for 20sec to speed up boot process" >> $fn
    echo "initial_turbo=20" >> $fn
    echo "force_turbo=0" >> $fn
    echo "arm_freq_min=250" >> $fn
    echo "core_freq_min=100" >> $fn
    echo "sdram_freq_min=150" >> $fn
    echo "over_voltage_min=0" >> $fn
  fi
  return 0
}

function enable_pwrsave {
# enable pwr save, LEDs off, HDMI port off, set cpu to lower freq
# echo 0 | tee /sys/class/leds/led0/brightness > /dev/null
# echo 0 | tee /sys/class/leds/led1/brightness > /dev/null
  /usr/bin/tvservice -o > /dev/null
  /usr/bin/cpufreq-set -g powersave
  _ret=$?
  return $_ret
}

function disable_pwrsave {
# disable pwr save, LEDs on, HDMI port on, set cpu to higher freq
# echo 1 | tee /sys/class/leds/led0/brightness > /dev/null
# echo 1 | tee /sys/class/leds/led1/brightness > /dev/null
  /usr/bin/tvservice -p > /dev/null
  /usr/bin/cpufreq-set -g performance
  _ret=$?
  return $_ret
}

ret=0
case "$1" in
      setup)
            setup_env "$2"
            ret=$?
            ;;
      start)
            enable_pwrsave "$2"
            ret=$?
            ;;
      stop)
            disable_pwrsave "$2"
            ret=$?
            ;;
      status)
            /usr/bin/cpufreq-info "$2"
            ret=$?
            ;;
          *)
  echo "Usage: ${0##*/} start | stop | status | setup [noswpkg]"
esac

exit $ret`

rvt, 2022-08-12:
What sort of power savings are we getting ?

chmorgan, 2025-02-07:
Closing due to inactivity. Changes like this are welcome, especially if they come with a report on how much power is saved over the baseline.

closed 2025-02-07 by chmorgan