stratux/stratux #189: Australia / NZ Flarm Frequency

flashtel, 2022-08-09:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x] no

    power source: Battery Pack

    usb cable:

3. EFB XCSoar 7.22

    EFB platform: OpenVario / Cubie Board

    EFB hardware:

4. Description of your issue:

I can see the following assets

[stratux-v1.6r1-eu028-32d5e58b-us.img.zip](https://github.com/b3nn0/stratux/releases/download/v1.6r1-eu028/stratux-v1.6r1-eu028-32d5e58b-us.img.zip)
558 MB
22 May 2022

[stratux-v1.6r1-eu028-32d5e58b.img.zip](https://github.com/b3nn0/stratux/releases/download/v1.6r1-eu028/stratux-v1.6r1-eu028-32d5e58b.img.zip)
558 MB
22 May 2022

I'm assuming the difference between them is the Flarm frequencies.

frequency bands: 
Europe/Africa SRD-F 868.0 - 868.6 MHz (frequency hopping), 
NZ 869.25 MHz, 
Australia 915 – 928 MHz (frequency hopping), 
USA 905 – 925 MHz (frequency hopping);

I've ordered [NESDR Nano 3] radios and tuned antenna for Australia Flarm / ADSB.

I can't find anywhere on how the frequency is set per country.

I also note the early flarm devices set there frequency based on GPS location, I don't need that but it's worth a mention.

What's required to update to AUS / NZ frequencies from the base image.

i'm good to modify config files command line, I did have a good look.

b3nn0, 2022-08-09:
This is not possible right now. The Demodulator is a binary blob provided by OGN that can't be set to different frequencies.

The difference between the "-us" build is only that by default, UAT is enabled, OGN is disabled, and some advanced settings are hidden to make it feel more like the "original" Stratux by cyoung, so it becomes a drop-in replacement for prebuilt units.

flashtel, 2022-08-09:
The OGN network receivers can be set to the Australian frequencies.

http://wiki.glidernet.org/list-of-receivers#toc3

I'm assuming it's the same binary in the receivers that you are using in your deployment. We have an OGN reciever at our club "ybev". I can't find much documentation on how to configure the OGN decoder for the different frequencies but it should be possible if you are using the same binary.

flashtel, 2022-08-09:
Mmmm, I should have done more research before I ordered the hardware.

http://wiki.glidernet.org/wiki:manual-installation-guide#toc7

That's how you configure the standard OGN receiver binary, is it possible to ask them? I understand the functional blocks and I have a unix background but I'm not a programer.

If there is anything I can do to contribute???

I've got your image running on a PI 3B+ at the moment. Radio's, GPS, Baro and Serial are in the mail already....

chmorgan, 2025-02-18:
Closing due to inactivity.

closed 2025-02-18 by chmorgan