stratux/stratux #197: GY-88 AHRS not detected

oliverstratux, 2022-09-07:
1. Stratux version: stratux-v1.6r1-eu028-32d5e58b.img
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [x] yes
      - [ ] no
      type: serial NEO M8N using pin 4/6/8/10 (working great)

    AHRS
      - [x] yes
      - [ ] no
     type: GY-88 (MPU6050 (accelero) addr 0x68 
                         HMC5883L (magneto) addr 0x1E
                         BMP085 (baro) addr 0x77)
      using pin 2/3/5/9

    power source: 220V to 5V DC 2A

    usb cable: 5V 2A rated

3. EFB app and version: SDVFR 1.276 / STRATUX for androïd 1.2.1 / firefox 192.168.1.10

    EFB platform: Android 9 

    EFB hardware: BLACKVIEW BV9800Pro

4. Description of your issue:

I would like to use my GY-88 as it have buil-in MPU6050 (similar to MPU9250) and BMP085 (similar to BME280)

I check I2C addresses with the 'sudo i2cdetect -y 1 method' so MPU6050 I2C address is 0x68 and BMP085 address is 0x77

I would like to change MPUREG_WHO_AM_I_VAL and BMPREG_WHO_AM_I_VAL
I'm a noob in Linux/Java and I'm not fluent in english so I didn't find the location of the 'root/stratux/main' folder following:[https://github.com/b3nn0/stratux/issues/115](https://github.com/b3nn0/stratux/issues/115)

Maybe it's due to my stratux version (v1.6 european) 

Is it possible to do a Stratux EU update to accept GY-88? (GY-88 is very common in DIY project)

Thanks in advance

BEMBELINIO, 2022-09-08:
I have the same problem to find the location of the 'root/stratux/main to change MPUREG_WHO_AM_I_VAL and BMPREG_WHO_AM_I_VAL address to 0x75 !!
Where is this sensors. file at ???
I am using latest stratux version May22 1.61 28....

b3nn0, 2022-09-08:
Feel free to try the beta build I provided here: https://github.com/b3nn0/stratux/discussions/182#discussioncomment-3161244
It should support the MPU6050.
However, the BMP085 is not supported. I'm happy to merge a pull request if someone wants to add support for it, but I won't do it myself - too busy right now.

The code does not ship on the Stratux. If you want to change the code, you need to clone this repository, make the changes, compile, push, and do a pull request. If you don't know how, you are probably not the right person to change the code  in the first place...

The steps are, just like so many other things, described in the wiki: https://github.com/b3nn0/stratux/wiki/Developing-Stratux

BEMBELINIO, 2022-09-09:
THX ! - b3nn0 for the qick response - WORKS!
the beta build update through the Webinterface made the AHRS work! 
[image]

- compass is not running , but this is because of the faked chip, I bougt on amazon, I suppose
https://www.amazon.de/gp/product/B07HMQZ7N9/ref=ppx_yo_dt_b_asin_title_o04_s01?ie=UTF8&psc=1
 ..but anyhow I `m glad becuse of the traffic function works stable so far with the baro
Great Job! - another saver Pilot up in the sky THX a lot - you are awesome !

closed 2022-10-27 by b3nn0