stratux/stratux #219: BME280 Pressure Sensor

rudiratlos, 2023-04-15:
1. Stratux version: 1.6r1-eu028
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: battery

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I have a GY-91 AHRS and Baro BMP280. The AHRS is working and recognized on the I2C-Bus, but the BMP has no address.
So I hooked up an 2nd board with an BME280 sensor and it show up on the I2C bus:
pi@stratux:/var/log $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 --  

How can I see that the Baro is used by stratux. Any logfile which cleary say, that the BME sensor is used.

b3nn0, 2023-04-15:
https://github.com/b3nn0/stratux/wiki/Altitudes-in-Stratux-EU#conclusions

closed 2023-04-15 by b3nn0

rudiratlos, 2023-04-16:
thanks, my GPS/AHRS page shows red -> no Baro is recognized.
Is the bme280 (not BMP280) supported. What could I do that my Baro can be used.
Install a special sensor overlay?

b3nn0, 2023-04-16:
Ah, sorry, I thought he BMP280 was the final one.
No, the BME280 is not supported right now (supporting it would be easier nowadays, as the "goflying" module is now also forked, but I don't have one to test).
See: https://github.com/b3nn0/stratux/issues/101

Are you able to compile Stratux yourself? If so, feel free to try what I suggested here: https://github.com/b3nn0/stratux/issues/101#issuecomment-774644691

If it works, let me know. I can then add it upstream.