stratux/stratux #225: Pressure altitude instrument blank in Foreflight

mattcheston, 2023-05-20:
1. Stratux version: stratux-v1.6r1-eu028-32d5e58b-us
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: u-blox 8

    AHRS
      - [X] yes
      - [ ] no

    power source: Anker PowerCore Essential 20000 PD

    usb cable: Anker

3. EFB app and version: Foreflight 15.4.1

    EFB platform: iOS 16.0

    EFB hardware: iPad Pro 11-inch 4th generation

4. Description of your issue:

    Pressure altitude instrument is blank in Foreflight. Pressure altitude appears correctly in Stratux web console.

[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

`http://192.168.10.1/logs/stratux/` was a 404 response. `stratux.log` is available here:

[stratux.log](https://github.com/b3nn0/stratux/files/11521325/stratux.log)

b3nn0, 2023-05-20:
Stratux reports the pressure altitude to FF correctly in its ownship report, don't know why Foreflight doesn't display that. Have you asked FF support why they don't do it?

rush0815, 2023-10-08:
Any news on that? I also don't get the pressure altitude shown on FF with my BMP280 sensor. On the Webinterface all is fine.

zyv, 2023-12-24:
I have the same problem on FF 15.11.1 on iOS 17 / iPad mini 6 during ground testing. Opened a ticket with FF to see if they are willing to look into it (1584319).

mattcheston, 2024-01-02:
I never figured this out. Appreciate others bumping the issue at the very least.

zyv, 2024-01-02:
Foreflight responded that they will try to have a look into it after the responsible team comes back from the holidays...

PapaGolf0422, 2024-04-18:
As of today 4/18/2024 the pressure altitude indicates and shows stratux as the source.  if you select cabin altitude, it shows the same data.  I think we can mark this one as fixed.

closed 2024-04-18 by b3nn0

zyv, 2024-04-18:
I can confirm this, apparently ForeFlight team fixed something. I'll let them know it worked.