stratux/stratux #240: error while installing stratux fresh.

fiamor, 2023-08-02:
1. Stratux version:
latest

4. Description of your issue:

when trying to start stratux by running this command 
`sudo make && make install && stxrestart`
on a fresh version of raspbian os, I'm getting 
`[cannot create regular file](cp: cannot create regular file '/opt/stratux//www/css/addtohomescreen.css': Permission denied)`
I fixed the double slash by changing line 10 of the Makefile under stratux/web to this 
`www=$(STRATUX_HOME)www`
but that wasn't enough to fix the issue.

`seraph@raspberrypi:~/Documents/stratux $ sudo make && make install && stxrestart
sudo apt-get install libncurses-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libncurses-dev is already the newest version (6.2+20201114-2+deb11u1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
sudo apt install librtlsdr-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librtlsdr-dev is already the newest version (0.6.0-3).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
git submodule update --init dump1090
cd dump1090 && make BLADERF=no
make[1]: Entering directory '/home/seraph/Documents/stratux/dump1090'
Building with:
  Version string:  unknown
  Architecture:    aarch64
  DSP mix:         aarch64
  RTLSDR support:  yes
  BladeRF support: no
  HackRF support:  no
  LimeSDR support: no
make[1]: Leaving directory '/home/seraph/Documents/stratux/dump1090'
git submodule update --init rtl-ais
cd rtl-ais && sed -i 's/^LDFLAGS+=-lpthread.*/LDFLAGS+=-lpthread -lm -lrtlsdr -L \/usr\/lib\//' Makefile && make
make[1]: Entering directory '/home/seraph/Documents/stratux/rtl-ais'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/seraph/Documents/stratux/rtl-ais'
make -C web
make[1]: Entering directory '/home/seraph/Documents/stratux/web'
mkdir -p /opt/stratux//www
cp -r css /opt/stratux//www/
cp: cannot create regular file '/opt/stratux//www/css/addtohomescreen.css': Permission denied
cp: cannot create regular file '/opt/stratux//www/css/ahrs.css': Permission denied
cp: cannot create regular file '/opt/stratux//www/css/dark-mode.css': Permission denied
cp: cannot create regular file '/opt/stratux//www/css/main.css': Permission denied
cp: cannot create regular file '/opt/stratux//www/css/map.css': Permission denied
cp: cannot create regular file '/opt/stratux//www/css/ol-layerswitcher.css': Permission denied
cp: cannot create regular file '/opt/stratux//www/css/ol.css': Permission denied
cp: cannot create regular file '/opt/stratux//www/css/radar.css': Permission denied
make[1]: *** [Makefile:14: all] Error 1
make[1]: Leaving directory '/home/seraph/Documents/stratux/web'
make: *** [Makefile:58: www] Error 2`

fiamor, 2023-08-02:
this is also reproducible when running `make optinstall`

b3nn0, 2023-08-02:
You need root permissions to write /opt. Can't do that as a normal user.

closed 2023-08-02 by b3nn0

fiamor, 2023-08-02:
I thought running it as Sudo took care of that ? also the folders and the files already exist under stratux/www/css.

b3nn0, 2023-08-02:
You ran "make" with sudo, but not "make install" or "stxrestart"

sorry, this is not an "introduction to the Linux shell" support forum. This will be the least of the issues you will be facing.
If you want something that works much simpler and much more reliable, use a Stratux base image and start from there. All the heavy lifting is done there. See https://github.com/b3nn0/stratux/blob/master/image/mk_europe_edition_device_setup64.sh