stratux/stratux #241: GDL39

Snowflake6, 2023-08-10:
1. Stratux version: Future
2. Stratux config: Any

    SDR
      - [ ] dual

    GPS
      - [ ] yes

    AHRS
      - [ ] no

    power source:
      - [ ] USB

    usb cable:
      - [ ] USB 3.0, Cigarette lighter adapter 3.0A

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
    EFB platform/hardware:
      - [ ] Garmin Aera 660

4. Description of your issue:

Support for GDL39 output.  See [Issue 75](https://github.com/cyoung/stratux/issues/75) in the original Stratux repo where work has been started.  There is an offer in that thread from the developers of iFlyEFB to partially fund this functionality.

b3nn0, 2023-08-10:
Personally, I don't have much use for it, but I'm happy to review and merge a PR, once it's finished. I do follow the thread there, but I don't have access to the in-development GDL39 code.

Let's keep tracking this there, not here, until a PR is ready.

closed 2023-08-10 by b3nn0

Snowflake6, 2023-08-10:
Sounds good.  It seemed like the other code was not being actively worked on so I wanted to try and transfer the focus to this branch as it appears to be the "main" branch now.  But this thread is now linked in the other one so there is a connection and hopefully we can get the code ported to this branch once it's done.