stratux/stratux #246: no OGN receiption

rush0815, 2023-09-02:
1. Stratux version: 1.6r1-eu029
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: Neo-8

    AHRS
      - [ ] yes
      - [X] no

    power source: 1A Wall adapter / Powerbank 10mAh

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Nooelec 1090 Mhz
    868 Mhz Antenna of amazon
    Nooelec 978 Mhz      

5. Description of your issue:

I have installed the new version from image on two different plattforms. RaspberryPi 4 and a RPi Zero 2W. On both I can't get any OGN traffic. 

my settings:

GPS = ON
978 Mhz (UAT) = OFF
1090 MHz .... = ON
868 MHz(ONG), ADS-L)  = ON
162 MHZ (AIS) = OFF
Internet (OGN APRS) = OFF (to be sure I am getting real air-to-air traffic, if so)
AHRS sensor  =  OFF   (I don't have one yet)
Baro sensor = OFF (I don't have one yet)
Ping ADS-B = OFF
OGN Transmission .... = OFF

Show traffic source in callsign = ON
further, all logs off at the moment. 

I am located in the near of Mainz/Germany, a high density area. So to propability to get some FLARM traffic is very high. But I cant see anything. When turing on  OGN APRS, I see exactly one FLARM target, and thats a Windmill 
This problem does not belong to any EFB since I even can't see any FLARM/OGN traffic  on the stratux WebUI.

Related to the antennas: For OGN the 868 Mhz antenna is needed. But if I connect the 978 Mhz antenna, the range loss on the stratux status page is less than with the 868 Mhz antenna. To be honest, I don't trust that chinese tuning ;-)
A few days ago I ordered a NanoVNA to check what they have done...we will see....

Well, but I don't have any OGN traffic with both antennas and both PIs.
And yes... GPS is fixed to 3D.

Am I something missing? Any config?

BTW: Could you please post here or anywhere in the wiki what the callsing prefixes like "ea" and "e" and so on mean?   

Kind regards, rush0815

b3nn0, 2023-09-02:
> Related to the antennas: For OGN the 868 Mhz antenna is needed. But if I connect the 978 Mhz antenna, the range loss on the stratux status page is less than with the 868 Mhz antenna.

You should still use the 868Mhz antenna. Maybe there is just more noise on 868Mhz, which the 978Mhz doesn't pick up as well. Of course, in that case, it also wouldn't pick up many signals.

How high are your noise levels?
OGN reception range is - depending on noise between around 1NM and 1NM. It is very common to not see any targets there. The fact that even with APRS, you only get one target confirms that there is just nothing close enough.
Feel free to compare to what you see here: https://live.glidernet.org/#c=49.91285,8.24935&z=11&s=1
it uses the same APRS stream. If there is something there within 1-2NM, you should certainly see it in the Stratux, even with APRS disabled.

You might also want to check http://192.168.10.1:8082 (or whatever you configured your Stratux IP to).
This is a raw status web page for the OGN receiver. But if you see noise levels, this software is running fine, and obviously receiving stuff.

rush0815, 2023-09-02:
[image]

all @ 0pkt/min.

b3nn0, 2023-09-02:
The high message count are communication with the stratux process. GPS info, status messages, ... Ignore it.
Your noise levels seem to be high. Try to work on that. Everything seems to be running perfectly fine otherwise. I will close this until you confirm that there are actually transmitters nearby. I am 99.5% sure that there just are none.

closed 2023-09-02 by b3nn0

rush0815, 2023-09-02:
You were right, I got some OGN traffic now. It is shown on the traffic map, but ForeFlight don't show it. Only ADS-B

DocTopi, 2023-09-02:
Hello 3bnn0,
rush0815 wrote he uses a NooElec 978 Mhz device.
May be the bad 878 Mhz reception a problem of an insuffizient calibration for 878 Mhz?
Please see: https://dross.net/aviation/how-to-and-where-to-start-with-stratux/how-to-calibrate-nooelec-nano-2-sdrs-to-868mhz/
What`s your opinion?

rush0815, 2023-09-02:
@DocTopi
No! I said that I tested both the 978 and 868 Antenna. I have already changed thanged the serial of the one Sdr stick to stx:868:33.  Calibration already done.