stratux/stratux #247: no FLARM traffic on ForeFlight

rush0815, 2023-09-04:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ X] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source:

    usb cable:

3. EFB app and version: Foreflight on IPad

4. Description of your issue:

My Stratux is showing  OGN traffc but I is not shown in ForeFlight.  I have enable callsign prefix so I am able to distinguish the traffic sources. At the moment I see only "ea"  on ForeFlight. The Stratux map on the WebUI show salso "pa" as green arrow. It is also moving so the target is actually "fresh"

Settings:
GDL bearingless target circle emualtion = ON
Show traffic source in callsign = ON
GPS/1090/868  = all ON
Owinship Mode = default to (F00000)

Stratux IP is 192.168.101.61 (because connected as client to my phone with internet conenction)

Anything else I have to take care of ?

Kind regards, rush0815

b3nn0, 2023-09-04:
When generating GDL90, Stratux doesn't care what the source of the traffic is. Therefore, I'm almost certain that the OGN targets are in the GDL90 stream as well. Can you please cross-check with a different app? E.g. "Enroute" is free and has decent GDL90 support.

rush0815, 2023-11-04:
You were right! FLARM simply has a limited signal strength so that I could not get any traffic on ground.

closed 2023-11-04 by rush0815