stratux/stratux #249: No network on LAN when using AP+Client settings

brownleeb, 2023-10-04:
1. Stratux version: 1.6r1
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source:  A/C

    usb cable:  N/A

3. EFB app and version: N/A

    EFB platform: N/A

    EFB hardware: N/A
4. Description of your issue:  When using the AP+Client mode on RPi 4, I am connecting to the remote AP as a client and receiving an address from network DHCP server, but cannot access stratux via either web or ssh.  I was able to access via the stratux AP, both ssh and web.  Through testing, I found that the arp table on stratux only contained the DHCP server addres, and as soon as I "pinged" from stratux to any other machine on the LAN, connection was immediately enabled, but only to that machine.  Each machine had to be pinged separately from stratux to enable traffic.
5. Connection to stratux AP (192.168.10.x) works fine
6. Perhaps an ARP problem, or stratux not able to figure out which interface (wlan or ap) to use.
7. Or, could just be unique to the wireless chipset on Pi4.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2023-10-04:
Are you sure? Stratux does indeed only use the ARP cache to determine where to send GDL90. And unfortunately, this cache is only filled once the respective device makes first contact. However, it doesn't matter in what way that contact happens. Ping, web interface access, ... Once any contact is made to the Stratux, it will deliver gdl90 to it.

However, web or ssh access is completely unrelated to that and should always work. If it doesn't, something on a lower level is wrong. The Pi's wifi firmware is unfortunately kinda buggy wrt. this mode, so the feature never really left experimental stage..
However, it did work somewhat fine for me when disabling wifi channel hopping on my home network. Accessing the stratux from other devices on the network worked without issues.

brownleeb, 2023-10-04:
I'm doing my best to remember when I last made a change to those settings.  I dont normally use the client mode much, I just hit the problem and decided to dig into it today.
I will try internet passthrough.
I was trying to access via LAN IP.  I ALSO was connected via AP by connecting to the stratux wifi SSID.  It was by ssh-ing to 192.168.10.1 that I was able to get in and ping LAN clients.
I realize its probably in the firmware somewhere, just happened on the 'fix' and figured I would post it.

b3nn0, 2023-10-05:
FWIW, I just tried it again, but can not reproduce the problem.
I did notice though, that only one wifi client seems to be able to connect to the Stratux-AP. Not sure why that is.. probably due to the mentioned firmware issues or something. Let me know if you find anything useful.

brownleeb, 2023-10-05:
I dunno.  It doesnt make any sense to me.  Giving up for now.
It happens every time I reboot.  Today, when I ping from a linux box (3 different RPi's) the ping does not get an answer, but it is received by the stratux (arp table shows).  Tried two different android and Win11, no dice.
I enabled internet passthru and rebooted.  It changed all the settings (including OGN, etc) back to default.  I set everything back up again and rebooted, cant ping.
I added a line to the root crontab yesterday to run a ping to the gateway at startup, so that may be what makes that work (I'm just a bit too lazy to do everything to take it out)
Cant for the life of me figure out the issue but just wanted to add it to the bugginess of the wifi.
Reminder, this is an RPI 4, if that matters (certainly the chipset is a little different).  Its the only model 4 I have, Only other thing I have available is an RPI 2b being used for Flightaware feeds.