stratux/stratux #251: Feature Request - ICARUS protocol

icarus747, 2023-10-18:
1. Stratux version: 1.6r1-eu029
2. Stratux config: N/A 

    SDR
      - [X] single
      - [ ] dual

    GPS
      - [X] yes
      - [ ] no
      type: Adafruit Ultimate GPS 3.1 USB-C (required change of gps.go to ttyUSB0)

    AHRS
      - [X] yes
      - [ ] no

4. Feature Request:
  If possible, is there a way to output pressure altitude on the raspberry pi serial interface (or usb serial) for a RS232 connection?  Multi baud rate selector would be great but 9600 as default.  Output the pressure altitude in ICARUS protocol: 
  ALT<sp>12345<CR>
  Where:
      ALT ASCII characters
      <sp> space (0x20)
      12345 altitude in feet (pressure altitude from AHRS)
      <CR> carriage return (0x0d)

b3nn0, 2023-10-19:
Can you please elaborate?
Stratux can already provide pressure altitude via USB Serial in NMEA format ($PGRMZ IIRC).
This seems to be a very niche protocol. Who uses it? What devices can accept it? Why would you use Stratux if all you need is a pressure altitude? Where is the protocol specification?

icarus747, 2023-10-19:
It would be used to send pressure altitude to a Garmin GTX-327 transponder. Icarus protocol is an older format (ICARUS Instruments 3000) that is still used by some transponders that was an upgrade from Gray code.  Using a laptop, usb to serial, putty I could send this message over RS232 baud 9600 1,8,1 (start bit, data bits, stop bit) over to the transponder.  It won't accept GDL90 or NMEA format so was wondering if it would be possible to, instead of sending NMEA out the serial, to have the option to send just the pressure altitude in Icarus format.   Page 147 in this Garmin document briefly describes the format. 
http://static.garmin.com/pumac/GNS530W_InstallationManual_190-00357-02_.pdf
Image of a GTX after receiving pressure alt from serial.  It needs to be updated every second or the reading will go back to dashes. 
https://i.imgur.com/UOvVly0.png

icarus747, 2023-10-20:
Also to clarify, it would be pressure altitude from the AHRS, not GPS.  I think $PGRMZ is GPS altitude in feet. from what I was receiving over UDP: "$PGRMZ,501,f,3*1F" where the pressure altitude in the Stratux GUI was different.

b3nn0, 2023-10-20:
Nope, PGRMZ provides pressure alt:
https://github.com/b3nn0/stratux/blob/d7d4be46ba50d3deda770ba723d7e1648ca20004/main/flarm-nmea.go#L424

I don't think there is a lot of use in implementing this protocol directly in Stratux. I find it quite brave already that you want to feed certified radio equipment with a cheap BMP280, but I feel like you would be the only one ever needing it.

I'd recommend just writing a small python script or something that connects to port 2000 and converts the PGRMZ message to your desired format. Should be simple enough and looking at your github, you should easily be capable to do that.

icarus747, 2023-10-20:
Thank you B3nn0,  I was initially going to use this for bench testing with 'simulated data' so to speak. With your suggestion of just using python I'm going to try and us a RPI Pico W to listen and send over RS232.  This way I can test with different vendor NMEA output over wifi.

closed 2023-10-20 by icarus747

icarus747, 2023-10-28:
FWIW I have created a working example of this with a RPI Pico W.  https://github.com/icarus747/PicoICARUS