stratux/stratux #252: Support of BMP390 pressure sensor

scheuss, 2023-10-28:
1. Stratux version: v1.6r1

4. Description of your issue:
I designed and soldered a shield which includes the pressure sensor BMP390, the ICM20948 and some more battery charge/discharge stuff. According to the Bosch website the BMP280 and BMP388 are not recommended for new designs. The BMP390 and BMP388 are almost the same, the newer BMP390 is more precise. The Chip ID has changed from 0x50 to 0x60 (Datasheet page 30) therefore it worked not right away. 
Modifying the line 22 “ChipId    byte = 0x50” to ““ChipId    byte = 0x60” in the file stratux/sensors/bmp388/registers.go and rebuild it solved the problem but it is not a nice solution.
Maybe a better fix in the next release would be nice and I guess in the future this sensor will be on more out of the shelf breakout boards.

Best regards,
Roman

rush0815, 2023-10-28:
Hello Roman,
would you share some information about your board?

Kind regards,
Rush

scheuss, 2023-10-28:
Hello Rush,

Attached two pictures of the first version of the hardware. The pressure
sensor is "U3" top left of the picture.
I can give you more information and I will answer to you github response
soon.
I saw on you github profile you have a '.de' website. I guess you speak
German?
Then I would explain the rest in German, that would make it easier. I am
from Switzerland.

Best regards,
Roman

scheuss, 2023-10-28:
Hi Rush,
I think there is a major step between BMP280 to BMP388. The sensors BMP388 and BMP390 are almost the same. I downloaded both datasheets and compared it with the website: https://www.draftable.com/compare 
works very well.
Because the sensor BMP388 is already in your software implemented, there is almost no change necessary.
Yes, the sensor is working with my shield with this 0x50 to 0x60 modification.

Sure, see the attached files. During flight I had the problem of bad power connection of the 12V ‘cigarette lighter plug’ which powers the stratux. If there is one short interruption of power, the stratux restarts. During this time no GPS is available and after powering up, I need to reconnect from skydemon to stratus. Not ideal during a flight. 
This is the first hardware version, some changes are already necessary.
Best regards,
Roman
[image]
[image]
[SCH_StratuxBoard_V11.PDF](https://github.com/b3nn0/stratux/files/13196037/SCH_StratuxBoard_V11.PDF)

wadoadi, 2023-10-28:
Hi Roman,    This sounds great, and thanks for the info re support of the BMP 390, I had a design for the AHRS with the 390, I didn’t know it would work with just an address change, so scrapped it I may have to dust it off! It would be great if the core software supported the 390 and it’s address.ThanksAdrianSent from my mobile device

scheuss, 2023-10-29:
Hi Adrian,
To be more precise, it is actually not the address which has changed. The respond from the sensor at register read (Register Address 0x00, CHIP_ID, default value 0x60) has changed. But yes, just minor changes.
Best regards,
Roman

b3nn0, 2023-10-29:
As always in here: happy to merge a pull request :-)

b3nn0, 2023-11-02:
I just pushed support for the 390, by working with either ChipId.. it's completely untested though, so please confirm if it works correctly.

scheuss, 2023-11-05:
I replaced just the two files ‘bmp388.go’ and ‘registers.go’ and build the project with the command ‘make && make install && stxrestart.’ After reboot, I go a P-Alt value in the GPS/AHRS page, so I think it is working fine. I also tried to build an image with the ‘mk_europe_edition64.sh’ file and the command ‘mk_europe_edition64.sh prod branch’ but that did not work. I got an image, the raspberry booted but the stratux does not start correctly. I am not sure how to build such images.

Thanks for adding the support of the BMP390

closed 2023-11-05 by b3nn0