stratux/stratux #259: uAvionix Ping USB support

stefanux2, 2023-11-06:
# uAvionix Ping USB support
Reporting the original issue from the US version: https://github.com/cyoung/stratux/issues/879

## Introduction
The attached patch will enable the uAvionix Ping USB support via USB port and also all MavLink compatible traffic reported devices.

The pinUSB device does not have an internal GPS and is featuring MavLink.

Officially more than 100 units are up and running worldwide.

## Setup
1. Stratux version: Stratux 1.6r1-eu029 by b3nn0 (EU)
2. Stratux config:

    SDR
      - [X] single using uAvionix Ping USB via USB
      - [ ] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X] no

    power source: 12V to 5V

    usb cable: 10 centimeters

3. EFB app and version: PingUSB with Easy VFR 4 + SkyDemon + Foreflight

    EFB platform: iOS 17

    EFB hardware: iPad Mini + iPhone 14

4. Description of your issue:
I'm using Stratux built with uAvionix PingUSB device. This patch is to add the MavLink protocol to the stratux in order to enable the USB-Serial PingUSB device.

## Devices
1. Ping EFB
> USB Serial based with 1090 ES Protocol (Currently supported by Stratux)
> https://uavionix.com/downloads/pingefb/uAvionix-pingEFB-ADS-B-receiver-quick-start-guide.pdf
2. Ping USB
> USB Serial based with MavLink Protocol (This patch)
> https://uavionix.com/products/pingusb/

## Protocol
### Traffic Report Message
- Packet Id=**246** from _uAvionix Ping Integration Guide.pdf_
- https://www.google.com/search?q=mavlink+uavionix+integration+guide
[image]

### MavLink
- https://mavlink.io/en/guide/serialization.html#v1_packet_format

## Compilation
### Environment

1. Stratux EU with latest version
2. Remount the Stratux in RW /boot/cmdline.txt
3. apt update and install all the development tools
4. Install Go compiler
5. Create Go Environement
6. Modify main/ping.go
7. Make and copy the executable

### Download Patch Ping.go
This patch was compiled overwriting the main/**ping.go** file.
[stratux-serial-ping.zip](https://github.com/b3nn0/stratux/files/13266288/stratux-serial-ping.zip)

## Features

1. Support uAvionix Ping USB connected via USB to enable MavLink protocol
4. Bearingless target emulation
8. RSSI Emulation

## Test
Tested with 20+ hours of flight at sea level up to 8000ft in Europe/Italy.
[image]

## Integration on mainstream
Best integration is to add as second protocol option like "PingEFB" and "PingUSB" switch and rename it as pingusb.go

### Improvements
1. Split ping.go to pingefb.go and pingusb.go
9. Test with multiple disconnect-reconnection of the USB
10. Improve the crosscheck with valid data field
11. Improve Altitude integration with Stratux
12. Improve RSSI for bearingless traffic

b3nn0, 2023-11-06:
Could you crate a proper pull request with this, so I can properly review it?

b3nn0, 2023-11-15:
Was implemented here: https://github.com/b3nn0/stratux/pull/261

closed 2023-11-15 by b3nn0