stratux/stratux #265: configuring serial ( /dev/ttyAMA0 ) gps device

paony, 2024-01-06:
1. Stratux version:   v1.6r1-eu029

    GPS
      - [x] yes
      - [ ] no
      type:    ublox 9 connected via serial to /dev/ttyAMA0   (not via usb)

    
Problem:
any gps connected to the raspberry pi serial port is just treated as an ublox8 device (gps.go line 321)
Autodetection would be difficult to implement for a serial device because at the moment autodetection is done by usb id detection
also reconfiguring the baud rate from 38400 to 115200 did not work out of the box - it looses connection. (doing it manually with pyubx2 works)

possible solution:
make serial gps parameters configurable via the config file.
Would it be an option to include something like that in your release?
Where the ublox level, possibly the target baud rate and the device file can be specified via the user config
I could help with some coding and come up with a pull request.
Otherwise I would just do a quick hack for myself.

b3nn0, 2024-01-06:
Sounds reasonable. Happy to review and merge a pull request, as long as it is more or less clean and doesn't break other stuff.

VirusPilot, 2024-01-07:
I would certainly support such change as well as a similar case happens with /dev/serialin USB devices (e.g. T-Beams with different u-blox versions).

paony, 2024-01-08:
I started to working on it ...

closed 2024-01-10 by b3nn0

paony, 2024-01-19:
I have strated working on a Wike GPS page and would like to include some images. Where should I put them?

b3nn0, 2024-01-19:
What I always do: just paste them somewhere in an issue or issue-comment, then copy-paste the Markdown/URL from there. The link will stay active even if the issue/comment is never actually created.