stratux/stratux #269: Run SoftRf on the raspi as a stratux child process   (feature proposal)

paony, 2024-01-20:
With a complete stratux (Rx/Tx) currently we need to have 2 independent pieces of hardware (raspi and t-beam) with 2 completely independent pieces of software. The reason is that stratux itself does not have any Tx capabilities yet.Those 2 systems are only loosly coupled.

SoftRf has a raspberry pi edition which can transmit Flarm, ogn, ... via an RF module coupled to the raspi via SPI. Currently both stratux SPI ports are unused on for a typical build.
I had a brief look into the softrf code and it seems that it allready has the capapility to get GPS and config data piped in. So it should not be to difficult to run softrf as a child pocess of stratux and feeding the (10Hz high quality) stratux GPS data into softrf.

there are many compatible RF modules on aliexpress, some of them can transmit up to 1000mW. And they are  cheap - price range from 2$ to 15$. (search for sx1276 or sx1262 on aliexpress).

This would be a step to a closer integration of the TX part into stratux and such an setup would have many advantages, while I see only 1 disadvantage:
6 wires have to be attached connecting the RF-module to the raspi GND, VCC, MISO, MOSI, CLK, RST (maybe chip select, but that might as well be hard-wired)

So what do you think about that?

b3nn0, 2024-01-20:
Regarding the general idea:
I don't really see an issue in the loose coupling of a micro controller and the PI. It's just a bit more modular like this.
And since we need a GPS chip anyway, we might as well use the USB port for something more fancy.
Note however, that this is just the most common way it is done, because Stratux can already do what you suggest:
The ogn-rx-eu module we use can communicate with a Lora GPIO HAT to transmit. Unfortunately, this feature was implemented for a few very specific use cases, so I'd need to figure out what the details are again (supported chips, pinout). It really needs a wiki article...

paony, 2024-01-20:
Didn't know stratux can allready do that.
Could you give a hint where to find ogn-rx-eu source code, please

b3nn0, 2024-01-20:
Due to the proprietary, encrypted nature of the Flarm protocol, it is unfortunately not open source for now (hope that changes once they adapt ADS-L...). I don't have the code myself. It is a binary blob provided by the OGN folks.

rvt, 2024-02-12:
@paony it's not allowed her in Europe and other countries to deliver 1000mW at these frequencies. At max were are allowed to do is 14dBm (which is not even close to 1000mW). That said, it's also pointless to deliver that much power if you cannot hear the other party because they are keeping legal restrictions.
There are forks of SoftRF that can deliver TX for OGN OR Flarm. Gx-Air-Com can deliver Flarm , but I am not sure if it can deliver OGN.. Both of them can supply GPS to Stratux

TheMiNuS, 2025-01-05:
@paony, you can find informations here on how to wire a SX1276 module to TX OGN frames. Look at how U5 is wired.

rvt, 2025-01-05:
Update, SoftRF from Moshe can deliver FLARM

paony, 2025-01-08:
Thanks for the recent updates. Actually I am using Moshes Softrf fork since a while on a china t-echo and like it (though I think range is not great). Currently I am not using stratux for transmitting at all. But I like the possibility to wire a SX1276. Where exactly is information about the wiring? Could not find it on a quick search, @TheMiNuS did you intend to send a link that got lost?

TheMiNuS, 2025-01-08:
You right my link was removed. Go to my repo and open the project StratuxDaughterBoard and open the schema in pdf format. You will have the reference to the famous U5 I was talking about.