stratux/stratux #274: Add u-blox M7 Power mode setting UBX_CFG_PMS

stefanux2, 2024-03-25:
1. Stratux version: latest EU
2. Stratux config:

    SDR
      - [X] single
      - [ ] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [X] yes
      - [ ] no

    power source: USB

    usb cable: +5V

3. EFB app and version: Easy VFR

    EFB platform: iOS Latest

    EFB hardware: iPad Mini

4. Description of your issue:

The request is to add the settings for UBX_CFG_PMS to switch between Default Power or Full Power.
The stratux set Full Power but on certain models of USB VK-162 the best is to keep it as "default" power.

Proposed solution:
- Add the management interface setting html
- Add the parameter to management interface
- Add default configuration to Full Power
- Add into gps interface the switch

Proposal source code will be posted by merge request during next days.

b3nn0, 2024-03-25:
Not sure I am a fan of it. It is a setting that 99.9% of users will not understand and shouldn't touch anyway. We should find a reasonable default that works. What gave you the idea that it should be changed? Can you elaborate a bit?

stefanux2, 2024-03-25:
> Not sure I am a fan of it. It is a setting that 99.9% of users will not understand and shouldn't touch anyway. We should find a reasonable default that works. What gave you the idea that it should be changed? Can you elaborate a bit?

After building many stratux with VK-162 USB dongle and testing them I noticed that commenting this line the time to fix and performance is much improved:
https://github.com/b3nn0/stratux/blob/0a5c01a5d8fae01e9062fa7bb7cc06ae82bd8b1a/main/gps.go#L569

I suspect that this kind of M7 works better with default UBX_CFG_PMS

b3nn0, 2024-03-25:
Surprising... But if that's really reproducible, than do that for all Ublox 7, not as a setting. I will conduct some tests tomorrow, too.

b3nn0, 2024-03-26:
Just did some tests with an M8N.
Testing methodology:
- Plug GPS in, start Stratux process to get our basic configuration on it, and wait for a fix, just so things are warmed up.
- Stop Stratux and connect via U-Center
- Trigger "Coldstart" and measure time to first fix to collect samples
- After 5 samples collected, switch to "Balanced Mode" and collect again

Result samples(seconds):
Full power 
47
47
88
59
53
-> Median: 53, AVG: 58,8

Balanced
51
54
50
53
30
-> Median: 51, AVG: 47,6

Admittedly, the sample size is not big, but I think we can safely say, that "Full Power" mode at least doesn't significantly improve anything.

Conclusion: We remove this setting entirely and use the default, assuming that the UBlox engineers know what they are doing.

commit 64edd25e (2024-03-26): Remove UBX-CFG-PMS configuration. Closes #274