stratux/stratux #277: Support for OGN-Tracker on T-BEAM with AXP2101 PMC

muccirob, 2024-04-04:
Stratux version: 1.6r1-eu029

The OGN-Tracker firmware version included in Stratux 1.6r1-eu029 doesn't work on the new T-BEAM boards mounting the AXP2101 power management chip, I know this is a known issue and tried to find a solution on pjalocha/esp32-ogn-tracker GitHub: reading the "issues" section on that project I wasn't able to understand if the problem has been solved by means of new libraries, but if so, recompile and generate the firmware with the new libraries is beyond my capabilities.
As all the new T-BEAM boards seem to be equipped with the new PMC, if someone could produce a new firmware release ready to be flashed would be a huge help for everybody in my same conditions...
Thank you!

b3nn0, 2024-04-05:
There is no such firmware yet unfortunately.
I will update as soon as there is.

There is a new project to replace the old OGN tracker repository, and it does work with the newer AXP chips, but it is still lacking so much functionality, that it is not reasonably useful to use with Stratux.
For now, I recommend staying with the other firmware options until it is ready.

b3nn0, 2024-06-17:
The new ogn-tracker now at least has ADSL and FANET support. I have pushed new firmware binaries and install scripts to the "pio-ogn-tracker" branch in case you want to do some testing and know how to install that. These are the variants I pushed: 
```
ogn-tracker-bin-tbeam07-sx1276.zip
ogn-tracker-bin-tbeam10-sx1262.zip
ogn-tracker-bin-tbeam10-sx1276.zip
ogn-tracker-bin-tbeam20-sx1276.zip
```

Unfortunately, the T-Beam S3 variants don't seem to build yet if ADSL is enabled.

closed 2024-06-29 by b3nn0