stratux/stratux #279: Major problem with T-Motion and GDL90

Bronkof, 2024-05-03:
1. Stratux version: 030
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source:

    usb cable: Rhaspberry pi

3. EFB app and version: Sky Demon

    EFB platform: Andoid

    EFB hardware: Android

4. Description of your issue: Signals of the t-Motion (legacy) protocoll are now allways displayed heading only north. Major problem when you exclude the hexcode from the t-Moton transmitter in the ownship hex list and you are using GDL90, even your own aircraft symbol ist only heading north and does not show speed or the time projection in front of the symbol. This makes it almost unusable.

Some people do not get Skydemon on the ipad connected by Flarm Airconnect so they have to use GDL90! A friend has this problem. So this ist a big problem for them.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2024-05-11:
Are you able to compile Stratux yourself from git?
I think it might be the same issue as https://github.com/b3nn0/stratux/issues/278
Unfortunately, I can't reproduce this either..

Bronkof, 2024-05-13:
Dear Adrian,

please send me an email to m.fani(at)gmx.net. I will send you my phone number so i can describe it more exactly. I can reproduce the issue any time! And i know other people wo have the same problem.

As far as i know i live close to you in south Germany.

b3nn0, 2024-05-14:
Fixed in 60df05a

closed 2024-05-14 by b3nn0