stratux/stratux #280: No USB devices recognized on Rasberry Pi 3B

mpotthoff, 2024-05-11:
1. Stratux version: 1.6r1-eu030
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: TTGO T-Beam

    AHRS
      - [x] yes
      - [ ] no

    power source: Power Bank

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

Hey! So after I unpacked my stratux again today and flashed the latest firmware, I suddenly had the problem that my raspberry pi 3B didn't recognize any of the usb devices anymore and only showed the root usb hub.

That appears to be caused by the changes of 474c031c82be9e792e2ebae4fccf16b9ebc37ad2.
When I searched for this online I found https://raspberrypi.stackexchange.com/a/138669. The solution provided there also helped in my case. I changed it to `dtoverlay=dwc2,dr_mode=host` and after a reboot all devices were recognized again.

b3nn0, 2024-05-11:
I kinda feared that this might cause problems.. but I am a bit surprised, it works perfectly fine for me and many others. What's special about your setup?

mpotthoff, 2024-05-11:
Good question... I really just have a stock Raspberry Pi 3 Model B v1.2 (at least that's what /proc/cpuinfo says) with only the latest stock stratux image flashed.

Someone else appears to have the same problem though: https://www.ulforum.de/ultraleicht/forum/2_technik-und-flugzeuge/8050_stratux-europe-edition/seite-131.html (FlyingRaki)

Vannue007, 2024-05-12:
I had to uncomment dtoverlay=dwc2, but that’s normal for a Pi zero 2W with usb hub

commit eb390c44 (2024-05-12): dwc2,dr_mode=host. Closes #280
closed 2024-05-12 by b3nn0