stratux/stratux #285: No WLAN when setting a secure (password protected) WLAN network in AP+Client mode

TheMZ2, 2024-06-02:
1. Stratux version: 
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [X] yes
      - [ ] no

    power source:
12V to USB converter

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
Skydemon
    EFB platform: (e.g., iOS 9.2)
IOS 15.8.2
    EFB hardware: (e.g., iPad Mini 2)
ipad mini 4
4. Description of your issue:
With the new release v1.6r1-eu030 of 7th of April Stratux started showing an abnormal behaviour.
The problem starts when in AP+Client mode i add a hotspot with a password and then set a password to the Stratux own WLAN.
As long as I don't set a password on Stratux WLAN network everything works as expected.
As soon as this password is set the Stratux stops providing the WLAN.
Apparently it keeps on working as it stays connected to the hotspot but I can't access the Stratux WLAN and I can't see what the thing is doing or even get a hold on the Logs.
I've tried rebooting the system but the only solution i've found is to reflash it.
Hope this gives you some hints on what went wrong.
Looking forward to a new release.
Thank you for providing the software for this mighty little system.
Cheers

b3nn0, 2024-06-03:
Unfortunately, I can't seem to reproduce that. Just tried to set it up, set a simple password (12345678) and enabled AP+Client mode.
It is now connected to my home network and offers an AP with the given password just fine.
Not sure what's going wrong for you.. any other info that might be helpful?

nikivan1, 2025-03-31:
I have the same issue with v1.6r1-eu032, on RPi Zero W2. The WiFi hotspot works fine until a secure connection is selected and a password is set. Stratux still shows in the available networks, but I get a connection failed error when trying to connect. If I go back to eu028, the WiFi secure connection works okay.

Hubbirob, 2025-05-25:
Same here (using v2.0-pre4). 

When choosing AP+Client Stratux stops providing a WLAN.