stratux/stratux #297: Missing mag. heading

Lupy720, 2024-09-30:
1. Stratux version: [Stratux Europe Edition 1.6r1-eu032](https://github.com/b3nn0/stratux/releases/tag/v1.6r1-eu032)
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: generic

    AHRS
      - [x] yes
      - [ ] no

    power source: independent

    usb cable: ---

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: ---

    EFB hardware: ---

4. Description of your issue:

Is the magnetic heading indicaton currently deactivated? I tried with the gy-91 board (with MPU9250 sensor) and with the ICM-20948 sensor. Both have a compass integrated, but the heading will not be displayed.

[image]

b3nn0, 2024-09-30:
Reading of the magnetometer is not implemented, and will not be implemented, as calibration in a mobile system is close to impossible.

closed 2024-09-30 by b3nn0

Lupy720, 2024-09-30:
I suspected that, but couldn't figure it out clearly from the old issues. 
Thanks for the clarification and the super quick reply!