stratux/stratux #298: First entry in Ownship Mode field not showing in EFB

pt1xoom, 2024-10-14:
1. Stratux version: V1.6r1-euo030
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: aircraft power

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Avarex 0.32

    EFB platform: (e.g., iOS 9.2) Android 14

    EFB hardware: (e.g., iPad Mini 2) Lenovo M9

4. Description of your issue:
I flew at a flight school and am a member of a flight club.  I have 8 aircraft in the Ownship Mode S/OGN Codes field.  I noticed a few weeks ago that the first aircraft in the list is not showing in my EFB.  I changed the order of the list and put that aircraft as the last entry of the list in the settings menu.  That aircraft showed up quickly in my EFB.  A few days ago, I was following another aircraft at the flight school and noticed it was not showing in my EFB.  Today I checked, and it is the first aircraft listed in that field.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2024-10-14:
The order of aircraft in the list has no relevance whatsoever.
The list is intended to do just that: don't show aircraft in the list as traffic, because you don't want to be alerted about your own presence all the time, so this is to be expected.
There is some logic in there to detect a "wrong" list, and the target couldn't possibly be you (far away/different heading etc). In that case, the aircraft will be shown despite being in the list. But this is certainly not perfect - it's just some crude guesstimation.

Yes, this "wrong-detection" can be used for lazy folks like me (and probably you) who switch aircraft often and don't want to reconfigure. But it will potentially lead to aircraft not showing. Not always, but it can certainly happen.
I tried to optimize this detection, but due to the variance in latency and position quality inherent to the system, it will never be perfect.

pt1xoom, 2024-10-14:
Ah.  Thank you very much for the explanation.

closed 2024-10-14 by b3nn0