stratux/stratux #307: T-Beam Flashing fails in multiple ways

Stefan-Kosker, 2024-11-22:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source: 5V 20000mAh Powerbank

    usb cable: Generic USB-MicroUSB Cable 7cm

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

When I try to flash the T-Beam, I get following stack trace error:

```
pi@stratux:/opt/stratux/ogn $ sudo /opt/stratux/ogn/install-ogntracker-firmware-pi.sh ogn-tracker-bin-tbeams3-sx1262-ubx
ogn-tracker-bin-tbeams3-sx1262-ubx.zip
Archive:  ogn-tracker-bin-tbeams3-sx1262-ubx.zip
  inflating: /tmp/bootloader.bin     
  inflating: /tmp/firmware.bin
  inflating: /tmp/partitions.bin     
  inflating: /tmp/esptool.py
esptool.py v2.7-dev
Serial port /dev/serialin
Connecting....
Traceback (most recent call last):
  File "/tmp/esptool.py", line 3065, in <module>
    _main()
  File "/tmp/esptool.py", line 3058, in _main
    main()
  File "/tmp/esptool.py", line 2757, in main
    esp.connect(args.before)
  File "/tmp/esptool.py", line 463, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/esptool.py", line 441, in _connect_attempt
    self.flush_input()
  File "/tmp/esptool.py", line 377, in flush_input
    self._port.flushInput()
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 588, in flushInput
    self.reset_input_buffer()
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 683, in reset_input_buffer
    self._reset_input_buffer()
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 677, in _reset_input_buffer
    termios.tcflush(self.fd, termios.TCIFLUSH)
termios.error: (5, 'Input/output error')
```

I own following device: U-blox MAX-M10S-00B, SX1262 868MHz, CHINA (https://de.aliexpress.com/item/1005005418286231.html - UBLOX 868 Variant)

[image]

what am I doing wrong? SOFT-RF Hotspot gets created and GPS Fix is also there so the device is fine

[image]

Am I using the wrong software? On chip, it says v3.0 and it is a U-BLOX variant so the only U-BLOX in the list is `sudo /opt/stratux/ogn/install-ogntracker-firmware-pi.sh ogn-tracker-bin-tbeams3-sx1262-ubx`. At the same time, the T-Beam UI shows a version 1.2 but the only 1.2 version to flash is with sx1276 chip, so it is probably not compatible neither.

I would like to try the other methods as well, but unfortunately the both links here are down and leading to a 404:
https://github.com/b3nn0/stratux/wiki/TTGO-T-Beam-Firmware-for-Stratux#flashing-from-linux

Help would be much appreciated.
Best Regards
Stefan Kosker
 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

VirusPilot, 2024-11-22:
you may have to put your T-Beam in Espressif Service Mode (press and keep holding **BOOT**, press and release **RESET**, thereafter release **BOOT**) before starting the flash script

VirusPilot, 2024-11-22:
did you really put it in Espressif Service Mode? don't mix it up with the UF2 upload mode

Stefan-Kosker, 2024-11-22:
> did you really put it in Espressif Service Mode? don't mix it up with the UF2 upload mode

I will be honest to you mate. If I knew what I was doing, I wouldn't be here. 

First time hearing uf2 upload mode and espif service mode. 

I will try again soon. How can I double check, if I am in service mode?

Stefan-Kosker, 2024-11-22:
It makes it hard to boot if the only buttons I have are power and rst btw. I got the same device as you suggested in your tutorials viruspilot. Did you had such an issue? Is the most right button boot button? I tried it with that as well but had no chance unfortunately

VirusPilot, 2024-11-22:
There is no confirmation about the service mode - only if e.g. you have booted SoftRF before and have an TX counter running, then in service mode this counter and the entire screen will freeze.

VirusPilot, 2024-11-22:
To force a power cycle, just remove the USB-C connector. Battery is anyway not recommended as it will prevent such a power cycle. Honestly, this device is quite picky sometimes ...

Stefan-Kosker, 2024-11-22:
So, usb out, finger on most right button, usb in?

I am asking this detailed, because as soon as it works, I will add a pr for updated readme

Stefan-Kosker, 2024-11-22:
Okay, so one step ahead:

```
ogn-tracker-bin-tbeams3-sx1262-ubx.zip
Archive:  ogn-tracker-bin-tbeams3-sx1262-ubx.zip
  inflating: /tmp/bootloader.bin     
  inflating: /tmp/firmware.bin
  inflating: /tmp/partitions.bin     
  inflating: /tmp/esptool.py
esptool.py v2.7-dev
Serial port /dev/serialin
Connecting...
Chip is ESP32D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x47)
```

interesting. I guess, I achieved the debug mode: This is the result of lsusb: `Bus 001 Device 006: ID 303a:1001 Espressif USB JTAG/serial debug unit`. I wonder why `Invalid head of packet (0x47)` occurs

VirusPilot, 2024-11-22:
Next option is to connect the device with your PC and flash it with Chrome: https://espressif.github.io/esp-launchpad/. I will try this option on my end as well and report.

Stefan-Kosker, 2024-11-22:
That is what I wanted to try before coming here. Sadly the link for flash is 404. Can I use the zipfile in the folder for that?

Stefan-Kosker, 2024-11-22:
Another question. Maybe I should try the 10-1262 firmware. After all, the board says v3.0 but the number after T22 is still a11 and according to docs, I should use 10-1262 firmware. What do you think?

VirusPilot, 2024-11-22:
> Another question. Maybe I should try the 10-1262 firmware. After all, the board says v3.0 but the number after T22 is still a11 and according to docs, I should use 10-1262 firmware. What do you think?

No, this package is for the ESP32 CPU. You need ESP32S3 versions.

VirusPilot, 2024-11-22:
> I will try this option on my end as well and report.

No success also on my side either. I have been using PlatformIO to flash the latest version with OLED enabled.

Stefan-Kosker, 2024-11-22:
I got something: 

https://esp32.com/viewtopic.php?t=21200

I think the package for our t-beam is compiled with an older version of esp-idf. 

@b3nn0 can you provide me with a small documentation what to compile for my esp32. I would create a pr when successful. 

Alternatively, another t-beam module? One without u-blox and v7 instead of V8 1262 chipset maybe? If it is more stable, I don't see any reason not to try it.

b3nn0, 2024-11-22:
The firmware binary is almost definitely not the problem, they are built correctly.
_maybe_ a too-old version of esptool.py (which is used for flashing) is. Unfortunately I can't really test that here.
In that case you should still be able to flash the firmware by other means.
I already fixed the download links, you can find the zip files here: https://github.com/b3nn0/stratux/tree/master/ogn

Stefan-Kosker, 2024-11-22:
> The firmware binary is almost definitely not the problem, they are built correctly. _maybe_ a too-old version of esptool.py (which is used for flashing) is. Unfortunately I can't really test that [...]

Will try later. Thank you.

Stefan-Kosker, 2024-11-22:
Or maybe, just maybe, we are doing a huge mistake here? 😅

https://github.com/b3nn0/stratux/blob/ebac60033aeebb82f08fa386408240aa18fa0b2f/ogn/install-ogntracker-firmware-pi.sh#L17

We define the chip as esp32. I will try it out with other chip and do a pr maybe.

b3nn0, 2024-11-22:
Could be. The flashing process for the s3 is not really tested tbh.
Keep in mind you need to stop the Stratux process to flash, otherwise you will get errors. See the script.

Stefan-Kosker, 2024-11-23:
Findings ( jump to next step if not interested):

removing --chip causes a return of a chip-hash, which is unknown to the esptool.py in the version 2.7 which is delivered in the zip file. esp-s3 chips are not supported until esptool.py 4.2 according to the forum

git pulling the esptool.py project (apt install has no global package for ubuntu and pip install not available beause of managed) and calling the function using it causes a  

```
  File "/home/pi/esptool/esptool.py", line 34, in <module>
    import esptool
  File "/home/pi/esptool/esptool/__init__.py", line 42, in <module>
    from esptool.bin_image import intel_hex_to_bin
  File "/home/pi/esptool/esptool/bin_image.py", line 16, in <module>
    from intelhex import HexRecordError, IntelHex
ModuleNotFoundError: No module named 'intelhex'
```

so, updating the version of esptool.py will not fix the issue. 

So the only way was using cx210 drivers and  esp32-s3 flasher software, which worked pretty clean. 

**Next Step:**

After the flash though, I can't confirm any working t-beam. The lsusb shows the device still stuck in flash mode I guess? 

[image]

But in Stratux status: 
[image]

What went wrong here in your opinion?

Stefan-Kosker, 2024-11-23:
Btw @VirusPilot  I wonder if it wouldn't be a smarter choice to abandon S3 chip and just buy the damn device suggested in b3nn0s documentation here 

https://de.aliexpress.com/item/4001287221970.html (sx1262 NEO-8 GPS module) It costs 50 bucks. So 30 cheaper than u-blox suggestion in your docs

VirusPilot, 2024-11-23:
Yes, the standard T-Beam is certainly good enough. The T-Beam S3 Supreme seems more suitable for standalone use ...

Stefan-Kosker, 2024-11-23:
> Yes, the standard T-Beam is certainly good enough. The T-Beam S3 Supreme seems more suitable for standalone use ...

Alright. I will order the suggested one from b3nn0s post and create a pr on your documentation than (on Monday, we have to go to a wedding today, ergo not available)

Stefan-Kosker, 2024-12-13:
With a T-Beam 1262 1.2 everything worked perfectly. Closing the issue. Thank you very much.

closed 2024-12-13 by Stefan-Kosker

VirusPilot, 2024-12-13:
@Stefan-Kosker What exactly should I fix in which of my documents? More then happy to do so ...

Stefan-Kosker, 2024-12-13:
The link in your purchase list targets a esp32-s3 device. This is not supported as we found out here. The one suggested in b3nnos documentation works (first device was dead on arrival though)

VirusPilot, 2024-12-14:
It seems that the T-Beam S3 Core module is no longer available separately. I'll make a note in my doc.