stratux/stratux #312: GPS BL1 disconnected after a while

mattia480, 2025-01-16:
1. Stratux version: v1.6r1-eu032 (ff1f01dc41)
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: GPYes BL1

    AHRS
      - [X] yes
      - [ ] no
      type: GY-91

    power source: https://aliexpress.com/item/1005005643325018.html

    usb cable: -

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: N/A

    EFB hardware: N/A

4. Description of your issue:

I am experiencing an issue with the GPS on my Stratux configured as follows:  
- Raspberry Pi 3B+ with 1GB of RAM  
- 2 x Nooelec NESDR Nano 2+  
- 1 x GPYes BL1  
- 1 x GY-91  

After a certain period of time (ranging from a few minutes to several hours), the application stops recognizing GPS NMEA sentences, even though these are correctly received by the Raspberry Pi (visible via `cat /dev/ublox8`).  
By simply restarting the Stratux application from the graphical interface (Developer > Restart Application) without changing anything else, everything starts working properly again.  

I have performed the following tests without success:  
- Verified that the power supply is sufficient using the command `vcgencmd get_throttled`, and everything appears fine (`throttled=0x0`).  
- Manually set the GPS type, serial port, and baud rate.  

Attached are also logs ([stratux.log](https://github.com/user-attachments/files/18437438/stratux.log)) and screenshots of the application and serial interface.
I am seeking support to understand whether this could be an application bug or a configuration issue.

Thank you.

[image]
[image]
[image]

b3nn0, 2025-01-16:
AFAIK `vcgencmd get_throttled` is only a momentary thing. Maybe run `sudo dmesg` after it happened and see if there are errors regarding voltage in the past.
I think the issue is somewhere in the lower levels (hardware).. Either power, or something is strange with your PI or your MicroSD is doing weird stuff or something.

I think that because your ogn-rx process seems to crash quite frequently too, with errors like this:
```
2025/01/15 23:02:22 OGN: ogn-rx-eu stderr:  r82xx_write: i2c wr failed=-9 reg=10 len=1
2025/01/15 23:02:22 OGN: ogn-rx-eu stderr:  r82xx_set_freq: failed=-9
```
indication some kind of communication issue.

mattia480, 2025-01-20:
Hi @b3nn0, thank you for your analysis. I confirm that I investigated the hardware functionality of the Raspberry Pi and discovered a malfunction in the USB port of the GPS. I replaced the Raspberry Pi with a new one, and everything is now working perfectly.  

I’d like to take this opportunity to ask why the noise value of the OGN differs between the OGN network information page and the colored bar on the Stratux status page, as shown in the attached images.  

Thanks!

[image]

[image]

b3nn0, 2025-01-20:
This is because ogn-rx does auto-gain-control to keep the background noise close to 0 (+1.3dB in your case).
However, always displaying 0 doesn't really help you when you try to debug noise issue. Therefore Stratux displays an artifical "Noise @ 50dB gain" value, so you can get a somewhat absolute noise measure, irrespective of what autogain does.

closed 2025-01-20 by b3nn0