stratux/stratux #356: Lilygo T-Beam V1.2

scheuss, 2025-03-03:
Hi,
It is getting really hard to buy the Lilygo T-Beam V1.1 which is needed for the Stratux with transmission capability. Does anyone have a good source of the V1.1, SX1262, 868Mhz, Neo-M8N?
The V1.2 does not work according: https://github.com/stratux/stratux/issues/277 
Is the OGN-Tracker firmware for the Lilygo T-Beam V1.2 already/soon available?
Does anyone know what the exact problem with the new version and its AXP2101 power manager chip is? If I use a V1.2 T-Beam, I measure only 2.6V on the VCC pin of the TTGO LORA module. Maybe it starts the LORA_VCC not correct. 
Schematic: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_TBeam_V1.2.pdf 
Has anyone an idea or a solution?

alexanderdross, 2025-03-03:
for the Lilygo T-Beam V1.1 SX1262, 868Mhz, Neo-M8N you may want to have a look [here ](https://dross.net/aviation/shop/t-beam-868mhz-for-stratux-flarm/)

Toogo, 2025-04-03:
I just bought a V1.2 with M8N on AliExpress, working great

scheuss, 2025-04-08:
Very interesting, I thought there is no working firmware for the new power manager IC.
I have the module V1.2, SX1262.
@Toogo Which firmware did you flash to the lilygo board? (ogn-tracker-bin-tbeam10-sx1262.zip, ogn-tracker-bin-tbeam10-sx1276.zip, ogn-tracker-bin-tbeam12-sx1276.zip)

Toogo, 2025-04-24:
I use the v2Pre version:
https://github.com/stratux/stratux/releases/tag/v2.0-pre4

fbbln, 2025-05-16:
Any further info how to flash a V1.2 board with SX1262 correctly?
Is there a new FW included in 2.0-pre?

alexanderdross, 2025-05-16:
@fbbln [https://dross.net/aviation/how-to-and-where-to-start-with-stratux/how-to-flash-a-ttgo-t-beam-esp32-from-lilygo/](https://dross.net/aviation/how-to-and-where-to-start-with-stratux/how-to-flash-a-ttgo-t-beam-esp32-from-lilygo/)

fbbln, 2025-05-16:
I'm not sure I understand. The first link goes to the list of TTGO FW files which can be installed from stratux, but they do not seem to fit. I tried to use "ogn-tracker-bin-tbeam10-sx1262".
The module boots but I get no fix and errors on the serial:
```
Radio: Tx: 0:258:108:0:110  Rx: 0:0:0:0:0  -115.5dBm 0 pkts 0.0 pkt/s 60.0s
$PFLAU,0,1,1,1,0,,0,,*63
$POGNR,1,,,,+60.0,,,*46
$GNGGA,142701.00,5228.95141,N,01318.79400,E,1,12,0.89,54.3,M,42.3,M,,*72
$GNGSA,A,3,02,01,32,03,08,17,22,14,10,,,,1.43,0.89,1.11*11
$GNGSA,A,3,68,84,69,67,75,76,85,,,,,,1.43,0.89,1.11*18
[1039556][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[1039557][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
```
Or what else?

fbbln, 2025-05-18:
Despite these error messages it seems to work now.
I get a good GPS solution (with an external antenna) and it transmits OGN ID.
Now I only need to find out how to setup it right in the stratux web ui.

Electr0Fi, 2025-06-29:
@fbbln : Now a month later, is the v1.0 SX12**62** Firmware still working on the v1.2 T-Beam? Does it integrate with Stratux properly?
I've also recently purchased the v1.2 T-Beam with the SX12**62** chip, not realising there was only official firmware for v1.2 with the SX12**76** chip.

scheuss, 2025-06-29:
@Electr0Fi : I have a LILYGO V1.1 and V1.2 on my table for testing because I have the same problem. I flashed a SD card with the newest Stratux firmware ‘Stratux Europe Edition 1.6r1-eu032’. I programmed then both V1.1 and V1.2 LILYGO boards with the command ‘sudo /opt/stratux/ogn/install-ogntracker-firmware-pi.sh ogn-tracker-bin-tbeam10-sx1262‘ as descripted in this wiki article:
https://github.com/stratux/stratux/wiki/TTGO-T-Beam-Firmware-for-Stratux 

On the old LILYGO V1.1 with the power manager IC AXP192 the voltages to charge the backup battery, GPS_VCC and LORA_VCC are all 3.3V.
The LILYGO V1.2 has another power manager IC AXP2101. The voltage to charge the backup battery is only 0.6V, the LORA_VCC only 2.6V. The voltage for the GPS module is correct at 3.3V.
The power manager IC AXP2101 on the V1.2 needs to be configured differently as descripted here:
https://lilygo.cc/products/t-beam-softrf?variant=43170155692213

I think this is the reason why the supply voltages are not correct. I will try to test it according to your suggestion

Electr0Fi, 2025-06-29:
@scheuss Interesting information. Make sure the you grab a fresh copy of the repo for the OGN-Tracker. Pawel pushed a quick hotfix today as the v1.2 SX1262 firmware wasn't compiling. I'm not sure if that's why it's been excluded from the [Stratux Europe Edition 1.6r1-eu032](https://github.com/stratux/stratux/releases/tag/v1.6r1-eu032) and the upcoming [v2.0-pre4](https://github.com/stratux/stratux/releases/tag/v2.0-pre4)... 

After the quick hotfix, I was able to get the firmeware to compile and have added a Pull Request to the Stratux master branch (Added ESP32 OGN-Tracker firmware flashing support for Lilygo T-Beam v1.2 SX1262 #382), so that the binary gets included in future releases of Stratux. However it would be good to have confirmation from you that it does acutally work.

scheuss, 2025-06-30:
@Electr0Fi I tried to make an image of your fork (https://github.com/Electr0Fi/stratux.git) to test it as a system. 
Therefore, I setup a fresh raspberry pi with Raspberry Pi OS Lite 64-bit, Debian Bookworm

I executed the following commands:
```
sudo -s
apt install build-essential git
cd /root
git clone --recursive https://github.com/Electr0Fi/stratux.git
```
In the folder image_build I run the bash script with `bash build.sh` (with the command `.\build.sh` as in README.md I get the error `bash: .build.sh: command not found`)

I get the following error:
```
...
...
Submodule path 'image_build/pi-gen': checked out 'b9e30f2e0e2557ae62bde141eb657f48a2c21dae'
Submodule path 'ogn/ogn-tracker': checked out '7e18d4a7536f9c2a6df0e61998ba40884168e3cd'
Submodule path 'rtl-ais': checked out '01b8c9962eff1ee402fb678c85a4f1390dc3bc71'
fatal: No names found, cannot describe anything.
./docker_run.sh "cd data && make dpkg"
./docker_run.sh: line 6: docker: command not found
make: *** [Makefile:171: ddpkg] Error 127
Error creating the stratux debian package: Returned 2
```

Can you give me a hint how to build the image or provide the image for testing?

Electr0Fi, 2025-07-01:
@scheuss That's really strange. I have no idea why the build is failing. I didn't touch any other files or folders.

It's unusual that the command `.\build.sh` is failing. There is an extra `.` before the `build.sh:` in your error message. Is there supposed it be?

Both errors seem to point to the missing "commands" or files that should be in their correct places.

I'll try and build it myself when I get home today.

In the mean time, in order to test if my firmware builds are working (regardless of this image build error). You can do the following:
1. Build the current working, stable version of the Stratux image.
2. Download and copy this file [ogn-tracker-bin-tbeam12-sx1262.zip](https://github.com/Electr0Fi/stratux/blob/master/ogn/ogn-tracker-bin-tbeam12-sx1262.zip) into this location `/opt/stratux/ogn/` on your Stratux (where all the other firmware binaries are).
3. Run this command `sudo /opt/stratux/ogn/install-ogntracker-firmware-pi.sh ogn-tracker-bin-tbeam12-sx1262`
4. Once the process has finished, unplug the T-Beam and plug it back in (or unplug the Stratux and plug it back in. But the T-Beam should lose power once to ensure correct initialization)

Let me know if that works.

Electr0Fi, 2025-07-02:
@scheuss Thanks for the testing. I just fixed the sub-folder issue and pushed to my git.

Regarding the `esptool.py` that file is not present in the .zip files that I have in my repo, that I directly copied from the stratux/stratux repo. It must be created at some point...

scheuss, 2025-07-03:
@Electr0Fi 
I wonder where that esptool.py comes into the zip file. I searched in the Stratux code for that but I don’t find anything.

@b3nn0 There is no esptool.py in the binary OGN zip file if I download it from the Stratux repo and extract it on PC.
If I flash the newest Stratux image on a SD card, boot it and unzip one zip file in the /opt/stratux/ogn folder there is a esptool.py file. In the install-ogntracker-firmware-pi.sh file, the esptool.py is necessary. Do you know where this file gets into the zip file?

I found the problem for the low backup battery voltage, I will request the change in the https://github.com/pjalocha/ogn-tracker repository.
Beside that, I still have a problem with the connection between LILYGO and Stratux. I get the following error in the logfile: 
```
2024/08/25 18:36:11 GPS - none of the baud rates worked, gps not connected ...
2024/08/25 18:36:11 GPS - serial port/baudrate detection err: GPS - Failed to detect gps serial baud rate
2024/08/25 18:36:11 GPS - device detected at serial port /dev/serial0, assuming this is an ublox device, configuring as generic ublox:
2024/08/25 18:36:11 GPS - chip: generic ublox, device: /dev/serial0, baudrate: 115200
2024/08/25 18:36:11 GPS - consider to configure this device manually in /boot/firmware/stratux.conf for optimal performance
```

I need figure out what the problem here is