stratux/stratux #362: v2.0-pre4 missing rtl_eeprom binary

slindj, 2025-03-28:
1. Stratux version: [v2.0-pre4](https://github.com/stratux/stratux/releases/tag/v2.0-pre4)
2. Stratux config:

    SDR
      - [x] single
      - [ ] dual

    GPS
      - [ ] yes
      - [x] no
      type:

    AHRS
      - [ ] yes
      - [c] no

    power source:  battery

    usb cable: 6" high current charging cable

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: N/A
    EFB hardware: N/A

4. Description of your issue:

When running sdr-tool.sh to set the SDR serial string in the command prompt, script throws error that says rtl_eeprom: command not found and the SDR serial is not updated. Output as follows:

######################################################################
#                     STRATUX SDR SERIAL TOOL                        #
######################################################################

#            Setting 978mhz SDR Serial Data                  #
 SETTING SERIAL:
 rtl_eeprom -d 0 -s stx:978:0

Answer 'y' to the qustion: 'Write new configuration to device [y/n]?'

/opt/stratux/bin/sdr-tool.sh: line 81: rtl_eeprom: command not found

Do you have another SDR to program?
     'Yes' will shutdown your STRATUX and allow you to swap SDRs.
     'No' will reboot your STRATUX and return your STRATUX to normal operation.
     'exit' will exit the script and return you to your shell prompt
1) Yes
2) No
3) exit
#?

arcticdba, 2025-05-17:
I have to echo slindj's issue - no trace of rtl_eeprom in /usr/local/bin.

arcticdba, 2025-05-17:
Sadly it is not anywhere on the install. I decided to format the card and reinstall just to make sure I wasn't dreaming, but no, no trace of the rtl_eeprom anywhere. To make absolutely sure I did the following after having logged in as pi:

```
sudo su -
cd /
find . -name rtl_eeprom
```

and there was no hits. From what I can tell, the binaries are simply not there.