stratux/stratux #372: GPS invalid checksum

thomay86, 2025-04-25:
Hi everyone,

I just updated my Stratux setup from version 1.6r1-eu028 (32d5e58) to 1.6r1-eu032 (ff1f01d). Besides an - I suppose unrelated - issue with the baro sensor ([https://github.com/stratux/stratux/issues/371]) I have troubles with the GPS (ublox M8N-0-10 chip on a GY-GPSV3-NEO board) now: 

After booting, I receive an initial GPS fix for my actual location, then it drops out and goes to 'not communicating'/'disconnected' state on the status page. The log typically looks like this:
```
2024/08/25 18:43:13 GPS - device detected at serial port /dev/serial0, assuming this is an ublox device, configuring as generic ublox:
2024/08/25 18:43:13 GPS - chip: generic ublox, device: /dev/serial0, baudrate: 115200
2024/08/25 18:43:13 GPS - consider to configure this device manually in /boot/firmware/stratux.conf for optimal performance
2024/08/25 18:43:22 GPS - successfully opened serial port /dev/serial0 with baud 9600   (Valid NMEA msg received)
2024/08/25 18:44:58 GPS error. Invalid NMEA string: Checksum failed. Calculated 0X82; expected 0X63 [the hex values differ]
2024/08/25 18:45:00 GPS error. Invalid NMEA string: Invalid checksum
```

With manual configuration enabled in stratux.conf (`"GpsManualConfig": true,  "GpsManualDevice": "/dev/ttyS0",  "GpsManualChip": "ublox",  "GpsManualTargetBaud": 38400 [also tried 9600 and 115200]`) the same issue of invalid checksum persists.

As I understand the GPS fix (more precise: the time information?) is needed for OGN to decode the FLARM protocol. Is this initial GPS fix sufficient or will OGN stop working some time after because of the missing GPS? I could live with the missing position data, since I use my tablet GPS for navigation but I want to make sure that OGN doesn't stop working silently in the background.

Thanks,
Thomas

Electr0Fi, 2025-07-03:
This is quite interesting. I have a Lilygo T-Beam v1.2 with a SX1262 module, that also has a M8N GPS chip. I'm currently running SoftRF on it (I haven't flashed OGN-Tracker yet). However, once the GPS acquires a fix, the OGN decode kicks off, but then my T-Beam dies - the red GPS light turns off and and GPS data is no longer sent to Stratux, at least that's how it seems in the interface.

thomay86, 2025-07-05:
@Electr0Fi : Hi, and thanks for replying to this issue. Interesting thought on the power consumption... I'm also on a Raspi 3B, also two NooElec Nano 3 SDR connected, the GPS module at question, and the baro. No further hardware. It's all attached to a powerbank capable of delivering 2 Amps.

I'm just wondering, if the GPS issue has always been there without me noticing, or if it just came up after the update. I've been flying very little recently, but I'll try to verify that OGN is still working properly - that's my only concern.

thomay86, 2025-07-14:
@Electr0Fi I finally managed to check today... Even with the SDRs removed I see the same issue: GPS down after initial fix. So I can confirm your finding that it's not power related. If I find the time I'll prepare a new image SD card with my previous version (eu028?) and check if the issue persists.