stratux/stratux #377: Stratux AHRS 2.0 and GY-91 with mpu9250 and bpm280

jacobneider, 2025-05-20:
I've tried a few Statux AHRS 2.0 and GY-91 Boards. I have also flashed the latest and pre-release images a few times.
I have not been able to get them to connect. I have attached my stratux log

[stratux (1).log](https://github.com/user-attachments/files/20320942/stratux.1.log)

ibu-bonda, 2025-06-18:
Likewise, I'm still noob-level. I'm trying to find where the file '**_sensors/imu.go_** ' is at in the directories structure. At least I could change the WHO_AM_I register to **!= 0x70.**

oliverstratux, 2026-02-17:
I use gy-91 for years. No problems. Pays attention to pinout to gpio, not pin to pin compatible. You need to have a ''cable interface''. Please note that some chinese Gy-91 may kill itself without warning has there badly protected, but other solution aren't any better.. I just buy severals and change it when it die. There very cheap..

oliverstratux, 2026-02-24:
It would seem that the GY-91 chips are suicidal because their voltage regulation stage overheats and is sensitive to static electricity. A little clear varnish increases the contact area with the PCB of the SMD components, allowing for better heat dissipation and EMC protection. Just take care to not bloc the hole of the barometric sensor. A thermal pad can also help, but it's much more complicated to install effectively because the heat must not remain trapped in the pad. The pad must make contact between two heat exchange surfaces.